victronenergy/venus #1057: CerboGX can't see my main Wi-Fi

aquapro, 2023-04-07:
Hello,

my minimal setup:

CerboGX Firmware 2.91 and after first test updated to 2.93 (both the same issue).
FritzBox 7490 Firmware 7.29
2 x FritzWlan Repeater 1750E Firmware 7.29

Wi-Fi 2.4GHz and 5GHz aktivated, both the same SSID. SSID visible/broadcasted like this "mainground".

But I can't see this network. I can see other networks but not my main "mainground".
I switched off 5GHz for testing, but same issue.
With other Wi-Fi devices i can see my main Wi-Fi without any problems.

Any idea?

BR
Aqua

askpatrickw, 2023-04-09:
This thread may offer you some help.

https://community.victronenergy.com/questions/82313/cerbo-gx-not-connecting-to-wifi.html

aquapro, 2023-04-11:
Thanks for the link, I had already found it. But that does not help me. I do not see the network at all, no matter what I tried, so I can not connect.