victronenergy/venus #1058: Please add to serial-starter.rules by default

shaneyake, 2023-04-10:
Hello,

Can you please include this by default? This will allow us to connect serial devices to NodeRed without having to SSH into each GX which is not possible for remotely. Mostly a problem when doing a software update.

```ACTION=="add", ENV{ID_BUS}=="usb", ENV{ID_MODEL}=="VE_ignore", ENV{VE_SERVICE}="ignore"```

I would really appreciate it.

Thank you

mansr, 2023-04-11:
Are these USB devices with a custom product string, or what?

shaneyake, 2023-04-11:
Yes, it is a custom product string. Theses are really easy to program into the FTDI chip via the FT_PROG tool from FTDI.
VE_ignore would be "Victron Energy ignore" but I am happy to use any string that by default gets ignored.