victronenergy/venus #1063: wifi on raspberrypi zero2w and rpi4

mpvader, 2023-04-19:
Current status is that the wifi doesn’t work there, amd that we have a branch to fix it, but needs some further fixing.

——
raspberrypis: update Bluetooth and WiFi Firmwares
- This adds support on Pi Zero 2W and fixes wifi issues on RPI 4.
- In the 2.90 update WiFi firmware did not support for all RPI Zero 2 W board revisions, see posts from early october here: https://community.victronenergy.com/questions/144249/no-wifi-on-pi-zero-2w-venus-os-v29016-and-14.html

——

the branch: https://github.com/victronenergy/meta-victronenergy/tree/mbath/RPIZero2W

mpvader, 2023-04-19:
The error with that branch is:

ERROR: mc:raspberrypi2:packagegroup-venus-1.0-r0 do_prepare_recipe_sysroot: The file /lib/firmware/brcm/brcmfmac43430-sdio.clm_blob is installed by both linux-firmware-rpidistro and linux-firmware, aborting
ERROR: Logfile of failure stored in: /mnt/ram/venuspublisherbuilds/69ba2b0e/1/mirrors/venus/build/tmp-glibc/work/raspberrypi2-ve-linux-gnueabi/packagegroup-venus/1.0-r0/temp/log.do_prepare_recipe_sysroot.2335047
NOTE: recipe packagegroup-venus-1.0-r0: task do_prepare_recipe_sysroot: Failed
ERROR: Task (mc:raspberrypi2:/mnt/ram/venuspublisherbuilds/69ba2b0e/1/mirrors/venus/sources/meta-victronenergy/meta-venus/recipes-packagegroups/packagegroup-venus.bb:do_prepare_recipe_sysroot) failed with exit code '1'

—-

Mark will look at it

henrik242, 2023-05-18:
In case it's of any help: On my Raspberry Pi Zero 2 W Rev 1.0 I <a href="https://community.victronenergy.com/questions/144249/no-wifi-on-pi-zero-2w-venus-os-v29016-and-14.html?childToView=192441#comment-192441" rel="noopener noreferrer" target="_blank">had to replace /lib/firmware/brcm</a> with the contents from the <a href="https://downloads.raspberrypi.org/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2023-02-22/2023-02-21-raspios-buster-armhf-lite.img.xz">Raspberry Pi OS Lite (Legacy) image</a> to get wifi/bt working, as well as symlinking <em>BCM.hcd</em> to <em>BCM43430B0.hcd</em>.

jhofstee, 2023-06-09:
Closing this as it should be working, it can always be reopened if it doesn't seem to work.

closed 2023-06-09 by jhofstee

henrik242, 2023-06-09:
@jhofstee Thanks! Is this in v3.10~2?

nmbath, 2023-06-09:
Yes this will be in a 3.10 beta release.

henrik242, 2023-06-09:
@nmbath Cool. Is it already in 3.10~2 or in an upcoming release?

henrik242, 2023-06-09:
Ok, I just this post 5 days ago: https://community.victronenergy.com/questions/210133/venus-os-v3102-available-for-testing.html

So that's not a beta release, I guess?

nmbath, 2023-06-09:
I missed that post ;) If the build sys it is fixed in 3.10~2 then it must be fixed! I do not see anything in the releases notes however. If you have ea Zero2w then give it a go and see!

nmbath, 2023-06-09:
Looking at the change history for the related branch, it is in 3.10~4.