victronenergy/venus #1067: generator start/stop small improvements

mpvader, 2023-04-28:
- Make the new warm-up and cooldown work for both ACin-1 and ACin-2; instead of as its now only when generator is wired to ACin-1
- Hide or disable the warm-up, cooldown as well as `stop generator when ...` settings when the vebus firmware version doesn't support it yet.

mpvader, 2023-04-29:
Comment from Kevin in ~37 thread:

> The generator warm-up and cool-down options do not appear unless one of the system AC inputs is set to generator. All other generator settings appear even of no generator AC input is defined.

mpvader, 2023-04-29:
Hey @izak , has the VRM Portal devteam already been told to add the warm-up and cooldown state to the VRM Portal?

See this screenshot, and also see the `Stopped` in the second screenshot, that is also wrong I'd think.

[image]

[image]

mpvader, 2023-04-29:
dataAttribute gaS (1166) is missing the new enums:

[image]

kwindrem, 2023-04-30:
I noticed /Ac/Control/IgnoreAcIn1 dbus parameter can not be read.

Also a question: can there be a conflict between this and the ignore AC in mechanism set via assistants in the Multi?

mpvader, 2023-05-09:
@vejpasop pls check Kevins question. Also I asked Izak to make sure the four new paths are documented on the public dbus wiki page. Depending on the answer to this: pls include info about this there as well. Also that they can’t be read. Which for me is all fine, as long as it works for us (whicj it does) and pecularities are documntd

vejpasop, 2023-05-09:
The `/Ac/Control/IgnoreAcIn1` dbus item is only available with Multi firmware version 502 and up. This path is there to instruct the Multi to ignore the AC input. The actual ignore status is available on `Ac/State/IgnoreAcIn1` and is affected by ignore AC in control from the following "modules"

- dbus item `/Ac/Control/IgnoreAcIn1`
- Grid code 
- BMS
- The ignore AC in mechanism via the assistants  

So all of the above can control the ignore AC in. These are ORed. So if there is at least one "module" telling to ignore. The AC input is ignored.

kwindrem, 2023-05-09:
This sounds perfect. My concern is clearing /Ac/Control/IgnoreAcIn1 would upset the other ignores and allow AC in to pass. That won't happen based on the above description.

As mpvader says ot being able to read /Ac/Control/IgnoreAcIn1 is OK as long as it's documented. It's just odd to see a dbus parameter not show the current state in dbus-spy. I'm also guessing a test for valid would fail but haven't tried it.

Thanks to all that have worked on warm-up and cool-down.

mpvader, 2023-05-09:
@izak / @jepefe pls check all above, and let me know what (if anything) should be part of v3.00

One reason to make some of it part of above might change setting names or paths, and then if we don’t change it now, we’ll have to migrate it later.

mpvader, 2023-05-12:
All that really needed doing is done.

I’ve moved this to v3.10 now since there is one more open item on the todo. Which we might decide to not do but thats for later.

for now, this is good enough.

mpvader, 2023-07-28:
closing; its good enough.

closed 2023-07-28 by mpvader