victronenergy/venus #1072: v3.00: Batrium SOC no longer works

mpvader, 2023-05-06:
[image]

https://community.victronenergy.com/questions/202458/venus-os-v30037-available-for-beta-testing.html

izak, 2023-05-10:
OK... this is much weirder than I thought. The BMS at this site stops talking to us in Venus 3.00. As in, running `candump can1` literally shows no comms other than the 0x305 and 0x307 we send ourselves. As a result, we no longer detect the battery.

The issue I suspected: the content of 0x305, which also carries the voltage/current/sense info since 3.00~26, is unchanged at this site and doesn't have anything to do with it.

VRM shows 23 other sites with a Batrium BMS, running v3.00~26 or later, working perfectly.

Doesn't look like a Venus bug. Looks like something specific to this site.

izak, 2023-05-10:
The BMS is totally flooding the Bms.Can port with traffic. 300 messages a second. A lot of the traffic has id 0x122, 0x123, 0x124. The majority is 0x124.

The result is that about half the time, the battery is not detected. If you reboot, or simply kill and restart slcand, it recovers.

I checked other batriums: They don't send that much data.

This is definitely an issue specific to the setup of this site.

Sent our Batrium contact an email to find out what can be done.

In the interim, nothing we can do in Venus, and not a bug that needs to hold back Venus 3.00.

closed 2026-02-22 by jhofstee