victronenergy/venus #1075: Incorrect paths in the Official Victron commadline docs

askpatrickw, 2023-05-07:
https://www.victronenergy.com/live/open_source:ccgx:commandline

I'll just start with that there are lots of incorrect file paths (all of them?) on this page.

Suggestion: Move those docs off that WIKI to a GitHub Repo and publish them [via Sphinx to Read The Docs](https://docs.readthedocs.io/en/stable/tutorial/) and the community can help maintain the docs.

mpvader, 2023-05-07:
Hi, thanks for that. Read the docs and Sphinx looks nice, but its also yet another thing, so I decided to move that documentation here instead:

https://github.com/victronenergy/venus/wiki/commandline---operational

I've updated all paths that I noticed were outdated, not that many by the way.

If you see any more, and would like to fix them yourself, I'm more than happy to give you access to update that wiki

closed 2023-05-07 by mpvader

askpatrickw, 2023-05-10:
Moving it to the Wiki and putting up the "moved" page is also great!