victronenergy/venus #1076: Multi RS: Add support for Generator start/stop

izak, 2023-05-09:
At the moment a number of "conditions" in the generator start/stop code gets its values from the VE.Bus device, and for that reason it doesn't work with a Multi-RS. One example would be the AC output load, inverter temperature, inverter overload, or stopping when AC becomes available again (this last one may need some discussion, will likely need some kind of firmware work too).

Medium priority, after other Venus 3.00 and other stop/start things are done.

Wholistic, 2023-11-03:
This limitation was just raised again by a community member here - https://community.victronenergy.com/questions/240657/multi-rs-generator-start-problem.html

I will also see that it is added to the RS documentation limitations chapter until it is resolved.

mpvader, 2024-02-02:
@ReinvdZee FYI I assigned this one to you; since you're now taking care of generator start/stop.

ReinvdZee, 2024-05-30:
With systems based on multiplus/quattro, generator start/stop works as follows:

1. Systemcalc stores the vebus service of the multi/quattro (multiple devices show up as one `com.victronenenergy.vebus` service) in `/VebusService`
2. Generator start/stop looks for the path `/Ac/State/AcIn1Available` in the vebus service. If its there, the path `/com.victronenergy.generator.startstop0/Capabilities` is set to 1, meaning warmup/cooldown is supported. 
3. If this is the case, the gui displays the warmup & cooldown menu:
[image]
4. The user may edit these values and generator start/stop uses these values to properly control `/Ac/Control/IgnoreAcIn` of the vebus service. 

With the Multi RS, it is a bit different. The Multi RS connects to the GX with CAN, so each Multi RS shows up as a service (`com.victronenergy.multi.socketcan...`). Furthermore, the Multi RS does not have the path `/Ac/State/AcIn1Available`, used by generator start/stop to set the capabilities. The path `/Ac/Control/IgnoreAcIn1` has been added in a [recent beta release](https://github.com/victronenergy/venus-private/issues/329). 

**Plan to add Multi RS support to generator start/stop:**

1. systemcalc reports all `com.victronenergy.multi..` services in a new path.
2. dbus-generator looks for this new path if `VebusService` is empty or points to a device which does not support ignoring AC inputs. 
3. If found, dbus-generator can use `/Ac/Control/IgnoreAcIn1` of all of the multi RSes in the list. AFAIC, we can also use the presence of `/Ac/Control/IgnoreAcIn1` to set the capabilities (i.e. we don't need `/Ac/State/AcIn1Available` in the Multi RS).  

This won't work properly if:
1. There are Multi RSes and Multiplus/Quattro devices on the system (but generator start/stop can set `/Ac/Control/IgnoreAcIn1` on _all_ compatible devices, if needed)
2. Only _some_ of the connected Multi RSes should ignore the AC input

AFAIK these aren't valid use cases. Correct?

@izak, what do you think (systemcalc wise)?
@mpvader fyi.

ReinvdZee, 2024-05-30:
Possibly unrelated, but I also came across [this](https://github.com/victronenergy/dbus-systemcalc-py/blob/master/delegates/systemstate.py#L128-L133). In case of multiple Multi RSes in the system, systemcalc will then report the state of only the first Multi RS, is that a problem?

@izak

ReinvdZee, 2024-06-25:
>List of conditions in generator start-stop

The condition "On loss of communication" also relies on VE.Bus only and will therefore also not work with the Multi-RS.

mpvader, 2024-10-02:
This is being picked up now, and will use acsystem instead of .multi services directly

ReinvdZee, 2024-10-03:
[generator start/stop doc](https://docs.google.com/document/d/1NohCfWdM3Aj-FIk6D3GXrWsDlFFxs05h/edit)
@mpvader, @izak,

Generator start/stop support for multi RS will use `dbus-acsystem` to interface with the multis. On a system with Multi-RSes, there will always be at least one acsystem service. The Multi’s are combined based on their N2KSystemInstance, set up in VC. The acsystem relays most of the individual Multi paths. 

### How is the vebus- or acsystem service selected
dbus-generator looks at the systemcalc path `/VebusService` to fetch the vebus service name. If that path is empty, it will try to find the acsystem path with device instance 0. It won't pick the one with the lowest device instance because that shifts control to the next lowest instance if the current lowest one disappears. 

### Warm-up & cool-down
generator start/stop also relies on the vebus system for the warm-up and cool-down feature. It uses `/Ac/State/AcIn1Available` of the vebus service to check if the multi can ignore its AC input, which is required for this feature. On acsystem, this path is not available, but `/Ac/Control/IgnoreAcIn<x>` is. 
I changed it such that generator start/stop considers warm-up & cool-down supported when `/Ac/Control/IgnoreAcIn1` is present (for both vebus and acsystem).

### Startstop conditions
Here is a table with the startstop conditions that depend on vebus, listed with the path they're using and a corresponding path on acsystem.
Condition | Vebus paths | Paths present in `acsystem`| 
------- | ------------|----
AcLoad | `/Ac/Out/[L1,L2,L3]/P` | Yes
Inverter Temperature | `/Alarms/HighTemperature` | Yes (recently [added](https://github.com/victronenergy/dbus-acsystem/commit/76574b36c1145abbfbc59524c645ee211bea569f))
Inverter Overload | `/Alarms/Overload` | Yes (recently [added](https://github.com/victronenergy/dbus-acsystem/commit/76574b36c1145abbfbc59524c645ee211bea569f))
Stop on AC1/2 available | /Ac/State/AcIn<n>Available or /Ac/ActiveIn/ActiveInput and /Ac/ActiveIn/Connected | Only `/Ac/ActiveIn/ActiveInput` is present, see below
Start on loss of comms | Comms are lost if one or more conditions failed to evaluate several consecutive times. | No additional paths needed. This condition relies on the successful evaluation of other conditions.

#### Stop on AC1/2 available
In the case of a vebus system, this condition uses the path `/Ac/State/AcIn<n>Available` if available. If not, it will check `/Ac/ActiveIn/ActiveInput` and `/Ac/ActiveIn/Connected`. However, this fallback is only implemented for “Stop on AC input 1 available”. (my best guess is that  `/Ac/State/AcIn<n>Available` was introduced with the Quattro, so there is no need for this fallback on AC2 condition)

The acsystem does not have the path `/Ac/State/AcIn<n>Available`, but it also does not have `/Ac/ActiveIn/Connected`. The path `/Ac/ActiveIn/ActiveInput` path is present and equals:
- 0 if AC input 1 is connected
- 1 if AC input 2 is connected (Reserved for the Quattro RS)
- 240 if none of the inputs is connected.

I'll implement this condition for both AC inputs as:

1. Use `/Ac/State/AcIn<n>Available` if available (will only succeed on vebus). Otherwise, 
2. Use `/Ac/ActiveIn/ActiveInput` and `/Ac/ActiveIn/Connected` in case of vebus, or
3. Use only `/Ac/ActiveIn/ActiveInput` in case of acsystem.

mpvader, 2024-10-11:
@ReinvdZee please update the top post to be specific on what we're going to do and up to date. 

I heard from Izak that the feature  "Stop genset when AC input is available" is not useful, since its only useful when having a quattro; that probably needs amending?

Wrt the path /Ac/ActiveIn/Connected for Multis and Quattros: its a deprecated path, so good to not use or ask for that for the RS.

closed 2024-11-14 by jhofstee