victronenergy/venus #1077: Generator start on SOC not canceling after grid returns

KW4NP, 2023-05-09:
Problem: Generator starts on SOC condition but doesn't finish bringing the system up to the end SOC before the grid returns. 
Grid drops afters hours or days and generator kicks on again although SOC is not below the threshold. This can happen repeatedly as the grid comes and goes or drops below the required voltage. Even more annoying is that if the grid returns immediately it triggers a generator not detected warning while the generator finishes the minimum run time (because of issue #1031 ). Users are very confused because the generator is "Running for SOC" when SOC is well above the threshold. See attached example from VRM.
[LBIMMFLTCAL_log_20230502-1243_to_20230509-1242.csv](https://github.com/victronenergy/venus/files/11430849/LBIMMFLTCAL_log_20230502-1243_to_20230509-1242.csv)

mpvader, 2023-05-09:
for you @jepefe , or Izak?

mpvader, 2023-05-09:
and: is this a bug in a new v3.00 feature or regression in v3.00, so needs fixing as part of v3.00?

or is this something old and can be fixed after v3.00.

and @KW4NP thank you for your report. I saw the other duplicate issue, understand that now and deleted it.

jepefe, 2023-05-09:
I will take it, I'm going to look into it now.

jepefe, 2023-05-09:
@KW4NP
Please, let me know the Generator start/stop SOC condition values. Or enable remote support in that system so I can check it myself.

KW4NP, 2023-05-09:
> and: is this a bug in a new v3.00 feature or regression in v3.00, so needs fixing as part of v3.00?
> 
> or is this something old and can be fixed after v3.00.
> 
> and @KW4NP thank you for your report. I saw the other duplicate issue, understand that now and deleted it.

Yes, sorry for the duplicates. Github kept returning an error but was actually submitting the issue.

KW4NP, 2023-05-09:
> @KW4NP Please, let me know the Generator start/stop SOC condition values. Or enable remote support in that system so I can check it myself.

Starts at 30% and stops at 94%. I believe I've seen it happen even after 94% was passed while the system was on grid power. This happens on multiple systems and it's very confusing, especially when the power is unstable. It continually starts the generator when the power is bad, which is very common in this area.

mpvader, 2023-05-10:
hey @jepefe : I think you said this was not on v3.00; right? Then I'll move it to v3.10

jepefe, 2023-05-10:
Hi @mpvader, Yes, I can confirm now that this not new in v3.00, also is not solved in v3.00. 

We need to decide if this is an issue or a feature.

mpvader, 2023-05-10:
ok, moved to v3.10; even if its a bug its not something that I'm going delay v3.00 over anyfurther.

Lets discuss further tmrw.

thanks!

jepefe, 2023-05-26:
Finished.
dbus_generator: [v1.6.2](https://github.com/victronenergy/dbus_generator/releases/tag/v1.6.2)

Ready to be added to the 3.10 ToDo list once it is open.

KW4NP, 2023-07-21:
This is still occurring. While grid is down the battery drops to generator SOC threshold and starts. If grid returns before the battery is at the stop threshold, generator will stop as expected because grid has returned. Battery will recharge from grid/solar until it hits ESS threshold as expected. However, the moment the grid goes off the generator will start, because the ESS threshold is lower than the generator charge threshold for operational reasons. Even a momentary outage will start and stop the generator. The only way to get the system back to normal is to let the system finish charging on the generator.

mpvader, 2023-08-22:
@jepefe anything we need to do here before shipping v3.10?

jepefe, 2023-08-22:
@KW4NP The SOC condition does not have a delayed start since is a quite stable value so there is no risk that the generator starts due to a spike. 
So in your case the generator will start as soon as the SOC start setpoint is reached and there is no grid.

Have you tried v3.10 release candidates? If not, please do it and tell us if the new behaviour improved your case.
We changed the behaviour to reset the conditions state once grid is restored so the generator will not start if the grid fails again and the SOC start setpoint is not reached.

jepefe, 2023-08-22:
The last comment from @KW4NP is a mix of the original issue that must be fixed already plus a lack of delayed start for the SOC condition.

@mpvader Nothing else to do for the original issue, unless you want to add a delayed start for the SOC condition.

KW4NP, 2023-08-26:
I don't think there's anything that needs to be done if the fix is coming out in v3.10. I'll see what I can do to give that a try.
Thanks!

jepefe, 2023-09-06:
Enabled delayed start and stop to dbus_generator v1.6.6

jepefe, 2023-09-06:
@KW4NP 
We have enabled delayed start and stop to the SOC condition so if you configure the proper delay the generator won't start on short grid failures as you described in your post from Jul 21st.

KW4NP, 2023-09-06:
Thanks for taking care of that! Our grid is very unreliable so I think this will help a lot.

closed 2023-09-06 by jhofstee