victronenergy/venus #1080: Generator start/stop: add service hour timer that counts down

mpvader, 2023-05-09:


kwindrem, 2023-05-09:
FYI, GuiMods includes a generator service interval. If any of that code would be useful, feel free to use it.

I used a counter that counts up (in the code) but is displayed as time until service (count down) in the GUI. This made it much easier to change the service interval and keep the accumulated time so far.

KW4NP, 2023-05-09:
This is something that would be very useful for many systems. Different generators have different service intervals. Does the software count the time that the AC input from the generator is present or the time that the autostart relay is on? Seems like in the past the total run time counter only counted time that the generator autostart relay is on, which isn't the same thing as actual run time (i.e. if the gen doesn't start or if it's started independently).

mpvader, 2023-05-14:
Yes.

mpvader, 2023-05-14:
And perhaps leave it counting down, so you can at a glance see how many running hours its too late.

and whenever <= 0, raise the alarm.

also after a reboot I mean

mpvader, 2023-05-14:
And lets make it a warning. Not alarm. Sry for the many msgs

jepefe, 2023-05-15:
I've finished implementing this feature in the script and gui. Please take a look to the screenshots and let me know if you see any improvement.
Otherwise this can be tagged and added to the todo.

**Main page:**
[image]

**Run time and service submenu has been added:**
[image]
[image]

**Service interval exceeded:**
[image]

**Warning triggered when service interval is exceeded:**
[image]

jepefe, 2023-05-15:
Hi @kwindrem , 
Your implementation served as a good inspiration, thanks!

> FYI, GuiMods includes a generator service interval. If any of that code would be useful, feel free to use it.
>
> I used a counter that counts up (in the code) but is displayed as time until service [...]

kwindrem, 2023-05-15:
You are welcome.

I would keep times in hours since that's what is shown on the actual generator time meter. Days is useful if the generator is running close to full time. Otherwise, an indicated days could turn into weeks.

mpvader, 2023-05-15:
Nice work @jepefe , looks very good. And I agree with Kevin: many people will calculate in hours when it comes to generator run time. Not days or weeks. Especially anyone having a hybrid generator where the generator runs almost never or very little % of the time.

jepefe, 2023-05-26:
Finished.
gui: [feat_service_interval](https://github.com/victronenergy/gui/tree/feat_service_interval/qml) 
dbus_generator: [v1.6.2](https://github.com/victronenergy/dbus_generator/releases/tag/v1.6.2)

Ready to be added to the 3.10 ToDo list once it is open.

mpvader, 2023-07-28:
@izak we need to finish up the vrm (and modbustcp for good manners) side of this.

mpvader, 2023-08-22:
@izak pls don't forget this one.