victronenergy/venus #1087: Allow Victron MPPT and Victron SmartShunt to talk to Venus OS via Bluetooth.

ErikThiart, 2023-05-20:
Please guys, I don't want to run messy cables all over the place, the SmartShunt and the MPPT can already talk to each other via the VE Network, please allow the Venus device ( Raspberry Pi4 with Bluetooth in this case ) to also talk to this network. 

I have the Pi to far away for a USB cable run, but well within range for Bluetooth connection.

Wholistic, 2023-05-22:
Hi @ErikThiart,

What I have heard is that if it is to be attempted, the bluetooth connectivity between devices and the GX will first come to products that ONLY have bluetooth and no physical interface (such as the Smart Lithium Batteries). 

It would also require an external Bluetooth dongle for the Cerbo GX due to the issue that disabled Bluetooth when the chip heats up (tho perhaps not an issue for your Raspberry Pi).

mpvader, 2023-05-23:
Note that recent Cerbos don’t have that overheat problem anymore. See v3.00 changelog, its buried in there somewhere, and the info will make its way to the manual ofcourse.

bluetooth comms will for sure come some day; but I can’t give a time line.

mdshw5, 2023-05-24:
I've just replaced most of the power equipment in our camper with Victron gear. While the GATT readings in the iOS app are extremely useful, I've just set up a VenusOS instance on a Raspberry Pi and would love to be able to integrate all of my data in a tidy LCD dashboard. Please count me in as hopeful that VE.Smart networking comes to VenusOS!

ErikThiart, 2023-05-24:
I appreciate all the replies guys, perhaps some "add on" module or something that fetches the data and can send it via MQTT to Solar Assistant or even Venus OS itself via it's MQTT / Node-Red implementation. 

Unsure what the best route is, I just know the Raspberry Pi has great Bluetooth, and the MPPT and SmartShunt both support Bluetooth, there has to be a way.

lavdnone2, 2024-05-23:
Wish I saw this before I got Cerbo GX with a nice Victron screen. Guess it goes back to the store till it can display wireless VE.Smart info