victronenergy/venus #1089: New setting: Maximum AC grid voltage whilst exporting.

ajaxharg, 2023-05-25:
I have 13kW maximum export allowance through my G99. This is fine in the spring  with a max grid AC voltage of 249V. But now I am finding that if I don't keep changing the maximum allowed feed in figure the system goes over 253V. It has been approaching 260V before I changed the settings.  The G99 over voltage limit is 262.2V (from memory). I don't really want such a high voltage affecting my household equipment.

Could we have an ESS feed in setting which allows user configurerd maximum AC grid voltage?  This would allow me to leave the maximum export at 13kW but not worry about high household voltages when all the PV and turbines in the area push the voltage up.

I currently have a node-red script constantly changing the maximum export figure but I imagine this setting would be useful for a lot of people as more large single phase systems get installed.  I love how easy it is to knock up extra functionality using node-red.  I am limited to between 7 and 8.5kW export to stay below 250V.

Thanks
Andy

mpvader, 2023-05-25:
Hi @ajaxharg thank you for the suggestion! I can't give any timeline on this; and some may say that its better implemented in the inverter/charger since than the response speed will be much higher.

I've noted it down on a list of ESS settings / outside of this issue tracker.

closed 2023-05-25 by mpvader

ajaxharg, 2023-05-25:
Hi Thanks for the reply.I’m never quite sure of the divide between teams.  Is the list you talk about for the team on the Quattro or do you know who is a good contact for suggesting it please. Many thanksAndySent from my iPhone

mpvader, 2023-05-25:
Hi, we don't keep a public tracker for the inverter/charger. I am a good contact for it; don't worry!

ajaxharg, 2023-05-25:
👍thanks 😁Sent from my iPhone