victronenergy/venus #1092: "DC Loads" tile does not show up as lon as ESS uses external meter

tuxedo0801, 2023-05-31:
My setup:

* 3-phase Victgron Multiplus II 48/5000 setup with ESS
* EM24 for grid metering
* CerboGX
* several Victron MPPTs
* a 48V battery (of course)
* an additional 3rd party "island" inverter connected to the DC bus bar

Config:
* Setting on CerboGX: System Settings -> Has DC System: Yes

Expected behavior:
* The DC load from 3rd party inverter is shown on a separate tile in VRM, names "DC Loads"

Actual behavior:
* There is no such tile, **UNLESS** I setup ESS to use "inverter/charger" for "Grid Metering"
* --> Problem: I don't want to switch to inverter/charger for grid metering. I'm required to stick to the EM24.

So, how can this be solved? For me, this looks like a bug.

tuxedo0801, 2023-05-31:
Things I checked:

MQTT topic `N/{...}/system/0/Dc/System/Power` contains always correct DC load value. Even if EM24 is set as grid metering. But CCGX and VRM do not show it unless I switch to inverter/charger metering for grid.

Benoit3, 2023-06-02:
Hello,
I've exactly same issue with : 
- mutiplus 2 ess 
- smartsolar mppt 
- em112 grid metering
- raspberry Pi + venus os V2.90
- some dc consumers and "Has DC system" setting activated
The DC Tile doesn't display. To have it I need to disconnect the EM112 and reboot the Raspberry
Benoit

GerhardLang, 2023-09-24:
Hello.
Same problem as tuxedo0801.

- 3phase and 1 phase systems.
- VRM and remote console
- with GUImods are shown, without not

Jalle19, 2024-12-16:
Can confirm that this is an issue. The only place the "DC system" tile is visible is on the GX device.

Haven't tried with VenusOS 3.5 yet though.