victronenergy/venus #1094: hub4control: Make an option so a Multi on only L1 doesn't compensate for power on the other phases

izak, 2023-06-02:
There is a feature in hub4control, since the beginning, that if you have a Multiphase system, but you don't have Multis/Quattros on all phases, the power used on the other phases is still accounted for and the system will attempt to cancel it out.

A typical example is a system with a Multi only on L1, but a 3-phase energy meter. If a large load starts on L2, the Multi on L1 will attempt to feed power into L1 to cancel that load on L2.

This happens even if you set Multiphase regulation to "per phase".

What is needed is to either stop doing this when per-phase is selected, or to add an option to not do this. Since there might be people using this feature, I lean towards the latter.

mpvader, 2023-06-02:
See also L3 #8934 ; from today. Once there is a proof of concept/beta, we need to get in touch and install it there.

izak, 2023-07-18:
After some investigation, I found that this case is already correctly handled. "extraLoad" is not compensated for when multiphase regulation is set to "per phase". The problem is actually stemming from this issue instead:

https://github.com/victronenergy/venus/issues/1103

The system in L3-8934 has a Fronius, and the test was done when there was lots of PV and the batteries were full.

So I am closing this issue in favour of the other one.

closed 2023-07-18 by izak