victronenergy/venus #1095: 12 hour time support in GUI

mdshw5, 2023-06-04:
First off - thank you so much for supporting the DIY community! I really can't believe that these systems work so well. As an use from the USA, it does bother me that the system time seems to only report in 24h/military format:

[image]

However, with some community modifications we can customize the flow dashboard to display time in 12h format:

[image]

I would be grateful if you could consider a system-wide setting for time display format in some future version. Seems like many of your users in the USA might also appreciate this as well!

jhofstee, 2023-07-02:
I am not that familiar with the am/pm thing. What would you expect entering a time should do?

kwindrem, 2023-07-02:
In the US (and maybe only in the US), time is often shown as a 12 hour value with AM or PM appended to the end. Midnight is 12:00:00 AM,  noon is 12:00:00 PM.

In GuiMods, I use Qt.formatDateTime(new Date(), timeFormat) to return the text string either as 12 hour or 24 hour time.

timeFormat is "h:mm ap" for 12 hour time and "h:mm" for 24 hour time.

mdshw5, 2023-07-02:
👆yes thanks @kwindrem for the succinct answer. It would be great (maybe only for US users) to have this time format display set globally in the GUI.

thepaulcooper, 2023-07-03:
Some British people prefer am/pm but most, if not all, can easily switch from one to the other.

jhofstee, 2023-07-03:
Yes, the displaying part is trivial, but what would you expect a UI to do when setting a time, use 24h format or a separate toggle for am/pm. That is less trivial since 13 p.m. should be disallowed etc.

mdshw5, 2023-07-03:
I believe setting a time could still be done in 24 hour format. Most people should be able to deal with that.

kwindrem, 2023-07-03:
Most people will never have to set the date and time (presence of an NTP server will do that automatically). In any case, I agree with mdshw5 that setting time in 24 hour format is fine.