victronenergy/venus #1097: Upgrade nodejs to latest v18

mpvader, 2023-06-04:
- current version we are shipping is v14, and that is EOL already: 2023-04-30.
- v18 is current LTS and EOL 2025-04-30. v16 is EOL 2023-09-11.; so if possibly skip v16 and go straight for v18. More on the releases here: https://nodejs.dev/en/about/releases/

current status is that recipes are available, it builds; builds are available as well, and only thing tbd before inclusion is to test it.

Swu files are here: https://build-feeds.victronenergy.com/venus/mva/nodejs/images/

They’ve been built from this branch:
https://github.com/victronenergy/meta-victronenergy/commits/mva/nodejs

dirkjanfaber, 2023-06-05:
Tested on 2 different systems with the [pre-release test flow](https://github.com/victronenergy/node-red-contrib-victron/blob/master/test/pre-release-test-flow.json) that I usually use for new node-red-contrib-victron releases. Also tested with some other flows and nodes that are regularly used in Node-RED. I did not encounter any issues.

The other thing to test is Signal-K. I've done some rudimentary tests on it and did not encounter any issues there either (but not being too familiar with Signal-K, I might have missed something there).

mpvader, 2023-06-05:
Now that testing is done, I'd like to put this in master. But you had a few improvements to make that I'd then like to include first?

jhofstee, 2023-06-09:
included in v3.10~5

closed 2023-06-09 by jhofstee