victronenergy/venus #1098: Replace Mosquitto with FlashMQ

mpvader, 2023-06-05:
We're going to replace Mosquitto + dbus-mqtt by [FlashMQ](https://www.flashmq.org/) + [dbus-flashmq](https://github.com/victronenergy/dbus-flashmq).

The differences:

- FlashMQ + dbus-plugin is much faster, publishing all topics (1500) happens in 10ms. In Python this was 3 seconds.
- FlashMQ + dbus-plugin uses far less TCP messages by having multiple of those per packet. It was 2900, and with FlashMQ its 179 TCP packets.
- FlashMQ + dbus-plugin uses far less CPU.

Some details:

- Mosquitto in Venus OS will be replaced by FlashMQ
- dbus-mqtt (the python service) will be replaced by dbus-flashmq
- mqtt-rpc remains in place. This is the other python service, the one that takes care of all remote functionality like the dashboard on VRM; remote firmware updates, and so forth
- The cluster in the cloud is now also Mosquitto, and will be replaced by FlashMQ as well; but thats a separate project for later; not handled in this issue.

Changes with regards to keep alive and retain:

- @wiebeytec please elaborate.

Todos for all local MQTT traffic:

- [x] ~Add FlashMQ + dbus-plugin into Venus OS~ - done per Venus OS v3.10~1
- [ ] Test guiv2 - in progress
- [ ] Change guiv2 wrt keep-alive and retain - done; [ready for inclusion](https://github.com/victronenergy/gui-v2/pull/369)
- [ ] Test html5 app - in progress, see `#venus-html5` channel.
- [ ] Change html5 app wrt keep-alive and retain
- [ ] Change configs to make FlashMQ handle local MQTT + websocket traffic

Todos for remote MQTT traffic
- [ ] Finish broker to broker connection (action @wiebeytec )
- [ ] Change VRM wrt keep-alive and retain
- [ ] Remove Mosquitto + dbus-mqtt completely.

mpvader, 2023-06-05:
Flashmq & the dbus plugin are included in the build. But don’t use any of the standard ports yet.

To make it be used:

in /etc/mosquitto/mosquitto.conf, change the websockets listener port from 9001 to something else.

in /etc/flashmq/flashmq.conf change the websockets listener port from 9002 to 9001

restart both (in the right sequence 🙂 ); and wait a bit after stopping Mosquitto, since it waits a while before the port becomes available.

mpvader, 2023-06-09:
@wiebeytec I've updated above `Further details` section, please review.

mr-manuel, 2023-06-10:
Would it be possible to add also username and password protection to flash MQ, if the GUI is password protected? Like Node-RED?

This would be very useful in external/shared networks, where no network protection is possible.

mpvader, 2023-06-11:
Hi yes for sure, thats on the list if things to work on before release.

mpvader, 2023-06-21:
Per v3.10~9, the dbus-flashmq plugin was modified to only send all topics when explicitly asked for that.

Rather than sending it automatically when going from sleeping to kept-alive state.

This makes it more obvious to app developers that things are wrong if they don't also first request a full keep-alive.

dbus-flashmq commit is this one:
https://github.com/victronenergy/dbus-flashmq/commit/55fa32a5ae1275682c1cc9e9499fd085b1528c93

mpvader, 2023-07-07:
Per Venus OS beta version v3.10~12 we made local websocket requests be handled by flashmq rather than mosquitto.

See recent changes in meta-victronenergy for details, ie the v3.10~12 and preceding commits: https://github.com/victronenergy/meta-victronenergy/commits/master

mpvader, 2023-07-22:
Per v3.10~19, its:

1883 - FlashMQ  (this is the default port for non-encrypted MQTT)
1884 - Mosquitto (this is left here just for now, will be gone later)
8883 - Mosquitto (this is the default port for encrypted MQTT)
9001 - FlashMQ (this is the default websockets port, non encrypted)

Of above, 8883 is the last one that needs migrating to FlashMQ:

@wiebeytec, can you tell me what options to use in the flashmq config file?

https://github.com/victronenergy/meta-victronenergy/blob/master/meta-venus/recipes-connectivity/flashmq/flashmq/flashmq.conf

and Mosquitto:

https://github.com/victronenergy/meta-victronenergy/blob/master/meta-venus/recipes-connectivity/mosquitto/files/mosquitto.conf

mpvader, 2023-07-22:
Correction, also @wiebeytec . When making FlashMQ handle mqtt trafic from port 1883, I overlook that that would also change  dbus-mqtt & mqtt-rpc over to FlashMQ. Which is too early. And rather than fixing that as well I just reverted it for now, per v3.10~20.

We'll better continue this once flashmq broker to broker connection is ready. And meanwhile, anyone that wants to test against FlashMQ wrt local MQTT traffic can use port 1884.

Per v3.10~20 is it:

1883 - Mosquitto (this is the default port for non-encrypted MQTT)
1884 - FlashMQ (allows testing FlashMQ + dbus-flashmq plugin)
8883 - Mosquitto (this is the default port for encrypted MQTT)
9001 - FlashMQ (this is the default websockets port, non encrypted, used by html5 app - amongst other things)

mr-manuel, 2023-07-22:
Would it be possible to also make encrypted MQTT of FlashMQ available on port 8884 for testing? This way we can also make sure that this works.

wiebeytec, 2023-07-24:
For FlashMQ TLS:

```
listen {
  protocol mqtt
  port 8884
  # The keys from /data/keys/mosquitto.key, but we are not root so can't read them.
  privkey /etc/flashmq/mosquitto.key
  fullchain /etc/flashmq/mosquitto.crt
}
```

Note the comment line. FlashMQ is started as `flashmq`, so can't read the `/data/keys/mosquitto.key`. Mosquitto is started as root and then drops privileges, so it could also read it when the key is owned by `flashmq`. However, currently that means making `/data/keys` owned by `flashmq`, which is weird, because the SSH host keys are there as well.

Another possibility is sym-linking to `/data/keys/mosquitto.*` and only making those files owned by `flashmq` (not the dir they are in).

We do need to keep using these certificates (as opposed to making new ones), because people have likely used the `mosquitto.crt` file in the calling clients as root CA (because it's self-signed).

mpvader, 2023-08-04:
As [discussed today - see link](https://victrondevelopment.slack.com/archives/C8LV2EU94/p1691137679034519):

- dbus-mqtt sends most or all topics using the retained flag, but we want to get rid of that, so:
- dbus-flashmq used to send only system/0/serial and /keepalive as retained, but that causes issues - too nice/complex, so:
- dbus-flashmq per now no longer does that

mpvader, 2023-08-22:
We're preparing v3.10 for release within a few weeks. In that release, we'll be shipping Mosquitto, not FlashMQ. In other words, all changes so far will be taken out for that release.

FlashMQ will be included again in v3.20.

But first we'll keep FlashMQ in there for a few more days, to see if latest changes now fixes everything / have a good build for the HTML5 devs & Qinetic to work with.

mpvader, 2023-12-05:
Update:

- Per today our cloud has fully switched from Mosquitto to FlashMQ
- Per Venus OS v3.20~32, also on Venus OS we’ve replaced Mosquitto by FlashMQ.

With above, its now end to end FlashMQ, including mqtt-rpc.

Remaining todos concern security wrt the local MQTT connection, and testing.

mr-manuel, 2023-12-05:
It would be interesting, if you can give us some insights on how much the CPU load is now reduced.

erichabg, 2023-12-17:
It looks like mosquitto-clients is missing in Venus OS V3.13. I think this is usefull also with FlashMQ, isn't it?
Up to now I startet dbus-mqtt from /etc/rc.local with `mosquitto_pub -I cerbogx_evu -t 'R/<id>/system/0/Serial' -m '' -h 127.0.0.1`.
This fails now, as mosquitto_pub cannot be installed by `opkg install mosquitto-clients`.
As a  workaround, I enabled it via nodered.

mpvader, 2024-02-08:
What is needed for having FlashMQ in the official release of v3.20 is done. I'll move this issue to v3.30 for the completion things.

hanzoh, 2024-02-27:
I updated from 3.13 to 3.21 three days ago and just noticed that I do not get any values on local MQTT anymore.
The bridge connection from Home Assistant was repeatedly closed again and again.
The last value arrived at 18:46.

Checking the logs in /data/log/flashmq, I see the oldest entry only from 19:02:
```
@4000000065de31d70f3e272c [2024-02-27 19:02:37.255] [ERROR] On signal 'PropertiesChanged' in dbus_handle_message: std::bad_alloc
@4000000065de31d7235b430c [2024-02-27 19:02:37.592] [NOTICE] Accepting connection from: address='192.168.1.12', transport='TCP/Non-SSL', fd=21
@4000000065de31d72361079c [2024-02-27 19:02:37.593] [NOTICE] Client '[Bridge ClientID='core-mosquitto.venus', username='', fd=21, keepalive=60s, transport='TCP/Non-SSL', address='192.168.1.12', prot=3.1.1, clean=0]' logged in successfully
@4000000065de31d723f56be4 [2024-02-27 19:02:37.603] [ERROR] Packet read/write error: std::bad_alloc. Removing client.
@4000000065de31d723f58b24 [2024-02-27 19:02:37.603] [NOTICE] Removing client '[Bridge ClientID='core-mosquitto.venus', username='', fd=21, keepalive=60s, transport='TCP/Non-SSL', address='192.168.1.12', prot=3.1.1, clean=0]'. Reason(s): std::bad_alloc
```

I disabled MQTT on LAN and reenabled it, afterwards the MQTT messages started working again.

wiebeytec, 2024-02-27:
Can you try Venus 3.22, released yesterday? There was a memory issue with dbus-flashmq that was fixed in that release. It's very likely the issue you're experiencing.

hanzoh, 2024-02-28:
Thanks, I did not see that there was another update.
Have installed it yesterday and will monitor flashmq behavior.

ak68-hub, 2024-04-20:
I can´t login with blank username/ password in Version 3.31-4 to my Cerbo-GX (with MQTT-Evplorer its perfect :))

This problem should be solved since 3.31-1 !?

Do I need any additional configurations ?

wiebeytec, 2024-04-20:
> I can´t login with blank username/ password in Version 3.31-4 to my Cerbo-GX (with MQTT-Evplorer its perfect :))

Are you saying that MQTT Explorer does or doesn't work? Are you talking about another client that doesn't work?

Since 3.31~4, empty usernames should be supported. When I do:

```
mosquitto_sub -d -t '#' -h <hostname> -u '' -P '' -V 5
```

or

```
mosquitto_sub -d -t '#' -h <hostname> -u '' -P '' -V mqttv311
```

It works.

Do you have output from your client and from `/var/log/flashmq/current` on the GX device?

ak68-hub, 2024-04-20:
Thank you,
Now, I CAN connect from Nodered to MQTT-host
... but no new virtual device is recognized [venus-os_dbus-mqtt-battery]
I don´t know why ???

dirkjanfaber, 2024-04-22:
Tested anonymous login via Node-RED both with the `zero_byte_username_is_anonymous` option set to true and false. Also tested this both locally and via a remote mqtt session from Node-RED. All 4 options function without problems (on systems running 3.31~4).

@ak68-hub : Not sure what you are trying to achieve. Lacking more context, I doubt that it is flashmq related and I suggest filing an issue on the repo for [venus-os_dbus-mqtt-battery](https://github.com/mr-manuel/venus-os_dbus-mqtt-battery) instead.

mr-manuel, 2024-04-22:
He already opened a discussion https://github.com/mr-manuel/venus-os_dbus-mqtt-battery/discussions/21. That has nothing to do with FlashMQ nor the driver. Seems like a config issue of the user.

ak68-hub, 2024-04-23:
HI DirkJan,
I was successful now, the mistake was the main-topic in  the venus-os_dbus-mqtt-battery script (topic with "/" at the beginning :( )
Thank you
Andreas

mpvader, 2024-04-23:
Good that it works now @ak68-hub ! For any next issue, please stick to community. Our github issue trackers are not for support.

——

I’m closing this issue, all has been done, except for the security related work, which is handled in a different issue or tracker.

closed 2024-04-23 by mpvader

ogurevich, 2024-05-31:
the flashmq dameon on venus OS 3.32 seems to accept any user and any password while anonymous is not allowed in the flashmq.conf

Is it possible to protect connection to MQTT with user/passwort ? 
 
`mosquitto_password_file /etc/flashmq/mosquitto_passwordfile
allow_anonymous false
thread_count 1
plugin /usr/libexec/flashmq/libflashmq-dbus-plugin.so
max_packet_size 16777216
client_max_write_buffer_size 4194304
expire_sessions_after_seconds 86400
include_dir /run/flashmq
zero_byte_username_is_anonymous true

listen {
  protocol mqtt
  port 1883
  allow_anonymous true
}

listen {
  protocol mqtt
  port 8883
  allow_anonymous false
  fullchain /data/keys/mosquitto.crt
  privkey /data/keys/mosquitto.key
}

listen {
  protocol websockets
  port 9001
}`

mr-manuel, 2024-05-31:
I think it would be better to post the issue here: https://github.com/halfgaar/FlashMQ/issues

ogurevich, 2024-05-31:
I agree. Meanwhile, I’ve found a workaround solution for myself: I only enable the SSL listener on the Cerbo GX (port 8883) and protect it with an x509 client certificate.

wiebeytec, 2024-05-31:
This would have been the correct place for the bug, had it been a bug. Adding `allow_anonymous false` doesn't do anything, because it's already the default. It's the code in `/usr/libexec/flashmq/libflashmq-dbus-plugin.so` that always approves the login.

But, you'll be happy to know that authentication support is coming. It will be a general password in Venus for network services, that will also be used for MQTT.

ogurevich, 2024-06-01:
Thank you so much for the kind message. From my perspective, this functionality is very important and its prompt implementation is truly like a dream. :)

lumikcz, 2024-06-03:
> But, you'll be happy to know that authentication support is coming. It will be a general password in Venus for network services, that will also be used for MQTT.

Hi, If I understand correctly, this will be a solution for current workaround (or, at least what previously worked with mosquitto config) where we manually edited the authentication in config file over SSH? That would be great, since manual edits to config files don't persist when installing updates.

wiebeytec, 2024-06-04:
> Hi, If I understand correctly, this will be a solution for current workaround (or, at least what previously worked with mosquitto config) where we manually edited the authentication in config file [...]

Correct.

lumikcz, 2024-07-18:
Sorry to re-open this thread. Based on this closed issue: https://github.com/victronenergy/venus/issues/1138, I read it that this should be already working in Venus 3.40 released on July 17, but I haven't noticed it in the firmware. I assume the password-related feature has not been released in 3.40? Just want to understand when this can be expected to work :)

wiebeytec, 2024-07-18:
#1138 was closed because there are other tickets. And indeed, the feature has been pushed forward, probably to 3.50.

lumikcz, 2024-07-22:
> #1138 was closed because there are other tickets. And indeed, the feature has been pushed forward, probably to 3.50.

Thanks for update on this, understood.