victronenergy/venus #1099: Update SignalK to latest

mpvader, 2023-06-05:
- [ ] Dirk-Jan to update the recipes and shrinkwrap and such, test a build;
- [ ] Matthijs to add it master and ask someone from SignalK community to test it for real.

Please check signalk-server itself, wanted version is v1.46.3.

But also check each of the pre-installed plugins.

mpvader, 2023-06-07:
Ok thanks. Why the set HOME commit?

dirkjanfaber, 2023-06-07:
The HOME variable is used by Signalk to find its configuration files .In line 13 of `/usr/lib/node_modules//signalk-server/node_modules/application-config-path/index.js` it does a:
```
  return path.join(process.env.HOME, '.config', name)
```

When undefined, it won't start and throws an error. This is the best fix I could think of.

mpvader, 2023-06-07:
Ok. Strange, since until now this wasn’t necessary and the thing did work. I’m trying to see if there is extra  testing needed or something similar.

tkurki, 2023-06-11:
Oh, dependency on `HOME` is a side effect of https://github.com/SignalK/signalk-server/commit/d91d1fd23305aac64f55209ebf4343852a6aefde. This is by accident, not by design, but  I think something we can work with. We could probably tweak the code to proceed without failing without `HOME`.

SSL support out of the box in SK Server was meant to promote https usage, but I think there are very few people using this feature. Proper https requires certificates, which is beyond Joe Boater anyway, so having this work "always" is not a priority.

mpvader, 2023-06-11:
Hi @tkurki , thanks for the quick reply.

I don't have any issue with the HOME env variable being required; I just wanted to understand where its (suddenly) coming from.

I know enough - thanks!

closed 2023-08-02 by mpvader