victronenergy/venus #1100: Increase Cerbo Temperature Sensor Range

Wholistic, 2023-06-07:
The temperature range for the sensor (LM335) is specified as from -40°C up to 100°C.

[LM335 Data Sheet](https://www.ti.com/lit/ds/symlink/lm335.pdf?ts=1617553681821&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FLM335%253Futm_source%253Dsupplyframe%2526utm_medium%253DSEP%2526utm_campaign%253Dnot_alldatasheet%2526DCM%253Dyes%2526dclid%253DCjkKEQjwx6WDBhCCg-us6M74rM4BEiQA5XTz2lauuY3HmUHIWaTP67ZMxjsAne0qlmBLpFDBpmIADf3w_wcB)

[End user reports](https://community.victronenergy.com/questions/84043/cerbo-temperature-sensor-rang.html) cerbo gx reports an error below -20 C.

When trying to monitor the freezer temperature it trips the alarm when the temperature drops below minus 20 so they can't use the alarm feature (false positives). 

Requests increasing the temp range of Venus to that allowed by the sensor

mpvader, 2023-06-08:
As per our data sheet:

Temperature range is -20C to +70C.

i’ll see of this is easily possible and of not close this issue

mpvader, 2023-06-09:
Opposite of what I expected, It appeared to be quite easy!

Fixed per Venus OS v3.10~7

closed 2023-06-09 by mpvader