victronenergy/venus #1102: Incorrect use of 'empty array of unsigned 32 bit int' as dbus null

wiebeytec, 2023-06-09:
Null on the dbus is an array of signed 32 bit int (`ai`). Like [defined here](https://github.com/victronenergy/velib_python/blob/77ab979e2278ae28bf9161394abffa4d6629feb4/ve_utils.py#L12). There seems to be some inconsistencies.

In developing the dbus-flashmq-plugin, I noticed that many empty arrays use the dbus type `au`, meaning 'array of unsigned 32 bit int' instead of `ai` (array of signed 32 bit int). This results in this inconsistency when I stick to the rules:

```
N/48e7da87942f/vebus/257/AcSensor/7/Energy: {"value":[]}
N/48e7da87942f/vebus/257/Ac/ActiveIn/L2/P: {"value":[]}
N/48e7da87942f/vebus/257/AcSensor/7/Power: {"value":[]}
N/48e7da87942f/vebus/257/AcSensor/2/Location: {"value":[]}
N/48e7da87942f/system/0/Ac/PvOnGenset/L1/Current: {"value":null}
N/48e7da87942f/system/0/Ac/Genset/Runtime: {"value":null}
N/48e7da87942f/system/0/Ac/Genset/ProductId: {"value":null}
N/48e7da87942f/system/0/Dc/Alternator/Power: {"value":null}
```

In ESS demo mode, these services use `ai`:

- hub4
- logger
- modbusclient
- platform
- system

These seem incorrect and use `au`:

- solarcharger
- vebus

In dbus-flashmq-plugin, I just render all empty arrays as null to work around it. That is what the current dbus-mqtt Python implementation does too.

jhofstee, 2023-06-09:
https://github.com/victronenergy/dbus-recorder/blob/master/play.py#L36

jhofstee, 2023-06-09:
The c code does this since 10 years:
```
	/*
	 * Dbus has no invalid type, "maybes" have been suggested but not in mainline,
	 * So sent an empty array if the value is unknown.
	 */
	if (!veVariantIsValid(var)) {
		DBusMessageIter array;
		DBusMessageIter variant;

		dbus_message_iter_open_container(itt, DBUS_TYPE_VARIANT, "ai", &variant);
		dbus_message_iter_open_container(&variant, DBUS_TYPE_ARRAY, DBUS_TYPE_INT32_AS_STRING, &array);
		dbus_message_iter_close_container(&variant, &array);
		dbus_message_iter_close_container(itt, &variant);

		return veTrue;
	}
```

jhofstee, 2023-06-09:
If you are interested about the "maybes":

https://bugs.freedesktop.org/show_bug.cgi?id=27857
https://gitlab.freedesktop.org/dbus/dbus/-/issues/25 if you are interested.

wiebeytec, 2023-06-12:
The C code then uses `ai` as well (correctly), even though it's not as explicit about using the `i` type on purpose.

closed 2023-06-29 by jhofstee