victronenergy/venus #1105: MultiRS: Firmware update breaks VRM data logging until reboot

jepefe, 2023-06-20:
After updating the firmware of the Multi RS its related values on VRM are not updated anymore until Venus is restarted.

Happens every time so it is easy to reproduce by just updating the product firmware.

jhofstee, 2023-07-02:
Unrelated, but we should check VE.Can for the same issue, it is on my long todo.. So I wouldn't mind if somebody else checks this.

mpvader, 2023-07-02:
I don't understand what you mean @jhofstee , the Multi RS can only be connected over VE.Can.

martinbosma, 2023-07-03:
@jhofstee You mean like disconnecting a VE.Can device/service when we see bootloader messages?

jhofstee, 2023-07-03:
oh and that might actually be related, but for sure we should look at VE.Can for this issue and I never updated a Multi RS.

izak, 2023-08-24:
This is fixed in Venus v3.10~23. It was related to the race condition that causes energy calculation to be lost if the AC input type is not known at the instant the dbus service comes up.

https://github.com/victronenergy/venus/issues/1126

closed 2023-08-24 by izak