victronenergy/venus #1107: v3.01: update translations

mpvader, 2023-06-21:
In v3.00, the translations were missing.

dirkjanfaber, 2023-06-22:
All translations have been done. Created the https://github.com/victronenergy/gui/compare/master...faberd/translations branch, containing the updated terms and translations.

dirkjanfaber, 2023-06-23:
One more remark. There is a difference with the previous update of the translations, where only the `.po` files where updated. As both the `.po` and `.qm` files contain the translations, both have been updated in my commit. The `.qm` files however have been generated from the `.po` files. As that is also done when building the gui, I believe that the `.qm` files should be either removed from the repository all together or they should be kept up to date (which is what I've done in the commit).

jhofstee, 2023-06-28:
no, the qm files should be updated from the po files, the po files are the temporarily ones, just to be compatible with poeditor. The*.po files can be deleted though...

jhofstee, 2023-07-21:
This is also in v3.01 now.

closed 2023-07-21 by jhofstee