victronenergy/venus #1111: Support Multi RS as a PV Inverter

warwickchapman, 2023-06-23:
Support Multi RS to be added to existing site with Cerbo + VE.Bus.  Ie. Add Multi RS as a PV Inverter (no battery on Multi RS).

"The Victron AC PV solution all along (just comes with free battery inverter) ... it should communicate at least as well as the competitors with our GX products." -- GS

"Price is not the only choice, you will be amazed how many people will choose blue simply because it's possible." -- Anco

mpvader, 2023-06-26:
thanks Warwick, but note that this is for some day for me. Its not a priority over various other (Multi RS-related) projects.

warwickchapman, 2023-06-26:
Understood - in line my comments on same, but noting comments of others.  Will be a good value add when it is possible.