victronenergy/venus #1113: Raspberry 3 A+ VRM Portal ID Missing

Head, 2023-06-30:
As already discussed here https://community.victronenergy.com/questions/79169/raspberry-pi-3a-vrm-id-missing.html?childToView=212556 this issue can also be pinned down more here.

There is an issue in this file/commit https://github.com/victronenergy/meta-victronenergy/commit/3ea22639de3f698acbbd461b14ab854950d75872 as it lacks the `raspberrypi,3-model-a-plus` identifier. Though I don't know why the 3 A+ identiefies as **B**, should be in the kernel version. Maybe https://github.com/victronenergy/meta-victronenergy/blob/master/meta-bsp/conf/machine/raspberrypi2.conf is just missing `bcm2837-rpi-3-a-plus.dts`?

jhofstee, 2023-07-02:
Closing, see https://github.com/victronenergy/venus/wiki/raspberrypi-install-venus-image. It is a community effort, not ours, so we won't fix this, so it will just linger around here forever. Patches are welcome though.

closed 2023-07-02 by jhofstee

Head, 2023-07-13:
Excuse the stupid question, I'm only a Linux enduser, never touched anything related to the Kernel. You've said:
> The linux version as used doesn't have a bcm2837-rpi-3-a-plus.dts

But isn't this the right file? https://github.com/victronenergy/linux/blob/v5.10.110-rpi-venus-3/arch/arm/boot/dts/bcm2837-rpi-3-a-plus.dts with that Linux version linked in here: https://github.com/victronenergy/meta-victronenergy/blob/master/meta-bsp/recipes-kernel/linux/linux-raspberrypi.bb

nmbath, 2023-07-14:
@Head I think the issue is testing and validation. While some of the changes might look simple, the people that can do the work do not have a RPI 3a plus to test with and validate. Therefore the community needs to provide clear and concise details of changes.