victronenergy/venus #1121: dbus-modbus-client crashing

jhofstee, 2023-07-21:
```
[39940.248913] CPU: 2 PID: 26637 Comm: dbus-modbus-cli Tainted: G           O      5.10.109-venus-11 #1
[39940.258184] Hardware name: Allwinner sun8i Family
[39940.262913] PC is at 0xb6a98830
[39940.266132] LR is at 0xb6a80630
[39940.269286] pc : [<b6a98830>]    lr : [<b6a80630>]    psr: 60000010
[39940.275619] sp : bea56458  ip : b6ab92ac  fp : 00000000
[39940.280868] r10: 00000000  r9 : b69dca0c  r8 : bea5651c
[39940.286165] r7 : 018a32ac  r6 : 018dbfe0  r5 : 018e1f18  r4 : 018eaee8
[39940.292696] r3 : 00000000  r2 : 00000000  r1 : 00000000  r0 : 00000000
[39940.299259] Flags: nZCv  IRQs on  FIQs on  Mode USER_32  ISA ARM  Segment user
[39940.306501] Control: 10c5387d  Table: 42ce406a  DAC: 00000055
[39940.312256] CPU: 2 PID: 26637 Comm: dbus-modbus-cli Tainted: G           O      5.10.109-venus-11 #1
[39940.321376] Hardware name: Allwinner sun8i Family
[39940.326111] [<c010d19c>] (unwind_backtrace) from [<c010a2c8>] (show_stack+0x10/0x14)
[39940.333863] [<c010a2c8>] (show_stack) from [<c06583b0>] (dump_stack+0x98/0xac)
[39940.341095] [<c06583b0>] (dump_stack) from [<c01342a0>] (get_signal+0x794/0x79c)
[39940.348496] [<c01342a0>] (get_signal) from [<c0109984>] (do_work_pending+0x100/0x550)
[39940.356326] [<c0109984>] (do_work_pending) from [<c01000cc>] (slow_work_pending+0xc/0x20)
[39940.364496] Exception stack(0xc39affb0 to 0xc39afff8)
[39940.369549] ffa0:                                     00000000 00000000 00000000 00000000
[39940.377720] ffc0: 018eaee8 018e1f18 018dbfe0 018a32ac bea5651c b69dca0c 00000000 00000000
[39940.385894] ffe0: b6ab92ac bea56458 b6a80630 b6a98830 60000010 ffffffff
```

This is with an EV charging station

jhofstee, 2023-07-21:
I leave it up to you guys who is going to fix it.

mansr, 2023-07-23:
Is this reproducible?

mpvader, 2023-07-24:
I think so, its seen on one system. I'll contact you in Slack.

mansr, 2023-07-26:
This is likely a previously seen crash somewhere in the glib/dbus native code. Debugging, let alone fixing it, is very difficult. As it is a rare occurrence with minimal impact (the service restarts immediately), further investigation seems like a waste of time.

closed 2023-07-26 by mansr