victronenergy/venus #1124: check Venus OS large packages for newer versions

mpvader, 2023-07-28:
Hi Dirk-Jan,

this is the first version of an aim to check that a bit more structurally on a regular basis. Can you check them when you're back?

| package | releases | our version | available version | comments  |
| --- | --- | --- | --- | --- |
| [signalk-server](https://github.com/victronenergy/meta-victronenergy/tree/master/meta-third-party/recipes-extended/signalk-server) | [releases](https://github.com/SignalK/signalk-server/releases) |   |    |   |
| [signalk-n2k-ais-to-nmea0182](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-third-party/recipes-extended/signalk-server/signalk-server/add-venus-os-preinstalled-plugins.patch) | [releases](https://github.com/sbender9/signalk-n2kais-to-nmea0183/tags)
| [signalk-venus-plugin](https://github.com/victronenergy/meta-victronenergy/blob/master/meta-third-party/recipes-extended/signalk-server/signalk-server/add-venus-os-preinstalled-plugins.patch) | [releases](https://github.com/sbender9/signalk-venus-plugin/tags) |
| [nodejs](https://github.com/victronenergy/meta-victronenergy/tree/8eab54a00df70a4a5c339a77a52eb8473e0fac1e/meta-alternatives/recipes-devtools/nodejs) | [releases](https://nodejs.dev/en/about/releases/) |
| [node-red](https://github.com/victronenergy/meta-victronenergy/tree/f1e2727abdf50e07d21594c6182124d7925a74a5/meta-third-party/recipes-extended/node-red) | [releases](https://github.com/node-red/node-red/tags) |

dirkjanfaber, 2023-08-18:
I am now retrieving this, which has been automated, so it can be periodically ran. Question is where the output should go then. 

| package                                                                                                                                                                                                          | releases                                                                                        | our version   | available version   | up to date         | comment   |
|:-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------|:------------------------------------------------------------------------------------------------|:--------------|:--------------------|:-------------------|:----------|
| [nodejs](https://github.com/victronenergy/meta-victronenergy/tree/master/meta-alternatives/recipes-devtools/nodejs)                                                                                              | [releases](https://nodejs.org/en)                                                               | 18.16.0       | 18.17.1             | :red_circle:       | LTS       |
| [signal-k server](https://github.com/victronenergy/meta-victronenergy/tree/master/meta-third-party/recipes-extended/signalk-server)                                                                              | [releases](https://api.github.com/repos/signalk/signalk-server/releases/latest)                 | 1.46.3        | 2.1.1               | :red_circle:       |           |
| [signalk-venus-plugin](https://raw.githubusercontent.com/victronenergy/meta-victronenergy/master/meta-third-party/recipes-extended/signalk-server/signalk-server/npm-shrinkwrap.json)                            | [releases](https://api.github.com/repos/sbender9/signalk-venus-plugin/tags)                     | 1.30.2        | 1.32.0              | :red_circle:       |           |
| [node-red](https://github.com/victronenergy/meta-victronenergy/tree/master/meta-third-party/recipes-extended/node-red)                                                                                           | [releases](https://api.github.com/repos/node-red/node-red/releases/latest)                      | 3.0.2         | 3.0.2               | :white_check_mark: |           |
| [signalk-n2k-ais-to-nmea0182](https://raw.githubusercontent.com/victronenergy/meta-victronenergy/master/meta-third-party/recipes-extended/signalk-server/signalk-server/add-venus-os-preinstalled-plugins.patch) | [releases](https://api.github.com/repos/sbender9/signalk-n2kais-to-nmea0183/tags)               | 1.3.1         | 1.3.1               | :white_check_mark: |           |

mpvader, 2023-08-30:
venus-signalk plugin has been updated. Rest has deliberately not been updated.

closing, and there will be a new one for v3.20

closed 2023-08-30 by mpvader