victronenergy/venus #1134: generator start/stop: add option to skip warm-up when start is triggered by overload

mpvader, 2023-08-28:
a screenshot is coming from a distributor who made something for that in node-red.

mpvader, 2023-08-28:
ps. this is a nice to have, it is not necessary to take any risks wrt bugs in v3.10 release

ReinvdZee, 2024-08-08:
@mpvader Is this still relevant?

If so, the idea would be to add a switch here to skip warm-up?
[image]

philipptrenz, 2024-08-08:
I think this is really valuable. Systems that are dependent on the generator assisting in high load situations could otherwise not use warm-up at all.

ReinvdZee, 2024-09-23:
I've added the setting:
[image]

If enabled, warm-up will be skipped when start is triggered by the inverter overload condition. The cool-down time and generator stop-time are not skipped.

Should we make the setting only visible when the vebus system actually supports warm-up? (The warm-up & cool-down menu also won't be visible if it doesn't). I think it makes sense to hide this setting when there is no warm-up to skip.

@mpvader, @philipptrenz?

mpvader, 2024-09-23:
yes perfect.

closed 2024-09-24 by jhofstee