victronenergy/venus #1145: Notify users about MK3 firmware update

vejpasop, 2023-09-14:
Currently about only 42% of installations have updated the MK3 firmware. This implies they are missing out on features.
Likely cause: The MK3 firmware available message is hidden away in the Menu's.

[image]

**Solution:** 
Add a  warning popup that show that the MK3 needs to be updated. When done as a warning, the display starts blinking, but without the buzzer beeping. 

Warning text: `#51 mk3 fw needs update, see Device list -> Inverter/charger entry`

**Todo:**

- [ ] Add warning to GUI 
- [ ] After deployment, check the percentage of updated installations again

mpvader, 2023-10-01:
Here is how the notification looks like

[image]

For in the manual, appended to the already existing error code list:

GX Error 51 - mk3 firmware needs update

When this notification shows, it is recommended to update the MK3-controller that is embedded into the GX device. To do that, go into the Device list, and then into the MultiPlus-Quattro or EasySolar listing. Here you will see this entry:

[image]

There is a small chance, our statistics show approx. 5%, that this update triggers a short system restart: your inverter/charger might turn off and on again.

If you don't see that message, the system is already up to date. Note that this manual update is only needed this once. It was made to be manually initiated because of the small chance of the inverter/charger restart. Future updates can be installed without sometimes resulting in the short restart.

mpvader, 2024-02-08:
Released as v3.11, a long time ago, I've asked @evandromr to check what the %es now are.

Back in sept 2023, only 43% of MK3s was updated.

mpvader, 2024-02-12:
Percentage of updated systems went from 43% to 69%.

closed 2024-02-12 by mpvader