victronenergy/venus #1163: Add a QML page to interact with Node-Red

pontusr, 2023-10-20:
When using the Cerbo GX with a touch panel it would be great to be able to interact with the Node-Red flows.

For this I suggest an additional QML page to be added. First iteration of this page could simply list labels and buttons that are defined by Node-Red using new custom Node-Red nodes. Over time this could be enhanced to include both more advanced QT UI elements and more advanced control of this page from Node-Red.

Some wishes:
- Add new QML page that lists UI elements to interact with Node-Red.
- Add new settings if Node-Red is installed and enabled:
  - Slider to enable a "dashboard" QML page to be displayed if Node-Red is displayed.
    - Slider to make this new QML page the default page to display at startup.
- Add new custom Node-Red nodes to node-red-contrib-victron:
  - Node to control the display of a Qt Label.
  - Node to control the display of a Qt Button and detect changes in state
  - Node to control a "panel" of labels and buttons such as those on the already existing overview page
  - Node that can "emulate" a device so that a "panel" on the already existing overview can be controlled from Node-Red and optionally link to the new QML page

mpvader, 2023-10-21:
Hi @pontusr yes that would be very nice indeed.

For now, we are tied up in running projects, including the new version of the gui. Thereafter, we can look into this and other things.

Or is this something that you would/could work on?

pontusr, 2023-10-21:
Hi @mpvader

I didn't look into the details of working on this but would be willing to put some effort into achieving this functionality. But I guess it's only worth the effort if this type of functionality isn't already planned for the new gui and that the effort could be ported over.

The custom Node-Red nodes seems like the easy part and I think I could contribute there. I haven't worked with Qt in quite a while so I would need to some research on setting up a build environment and having a look at the level of abstraction of Dbus connectivity in the Qt parts before knowing if I could work on this. Would it even be possible to achieve this without the internal Victron code? If so a quick outline from you would help greatly.

aaronsb, 2023-10-24:
@mpvader Is there a GUI roadmap that would help align interests and actions? This way we can get a jump on possible Node-Red development value.

mpvader, 2023-10-26:
Hi again,

The roadmap is that I hope to this year replace the current gui with what for developers is called guiv2.

guiv2 is a project that has been running for a long time, and now nearing completion. Its a new design and all is reimplemented. Qt6 instead of Qt4 and WASM instead of VNC for remote console.

Keep an eye out in the  beta versions, we’ll announce how to test that in November or December.

For now its closed source, soon it will be open source. Meanwhile if you want I can give you access, it has documentation wrt build environment,

pontusr, 2023-10-26:
@mpvader Sounds exciting! 

For the particular feature I'm suggesting here I think it makes sense for me to test delivering something using the now current and open source version. I have some test Node-red nodes running already and will install Venus on a Raspberry here and try to get a Qt4 page working with that. I think such an effort should be easy to port over to the new version if my POC ends up working.
It is my current understanding (and hope) that I should be able to achieve this without compiling Venus from source. So for now I'm good without the new build environment.

mpvader, 2023-10-26:
hey @pontusr ok good.

note that for both the current gui as well as guiv2 you won't need to compile Venus OS from source.

And the current gui is not opensource either, except for the QML files which happen to end up in the rootfs (to great success of many people modifying them 👍 ). 

Anyway, sounds like a good plan to me; but if somehow you get stuck in that gui, don't spend too much time, since all work will need to be done again most likely.

The other thing I thought about is that I know that someone else is or will be working on adding (digital) switches to the gui; so there might be some overlap there.