victronenergy/venus #1168: can-bus-bms: Remove hardcoded DeviceInstance 512, allocate dynamically

izak, 2023-10-25:
At the moment, this hardcoded DeviceInstance means only one CAN-bms battery can be accommodated, even if you have two capable CAN-bus ports.

Dynamically allocating it will allow a second battery to go on 513.

p0l0us, 2024-08-31:
Many people including me are suffering by this issue. Is there any chance to fix this ? 
I'm happy contribute if anyone point me to right place.

We see two BMS connected via different CAN busses in the GUI, but there are issues - VRM is confused, system battery & DVCC controller is assigned randomly after CerboGX restart, also in node-red are values alternating within single battery. On other hand SignalK sees batteries separate with same `Path` (electrical.batteries.512), but different `Source`.

p0l0us, 2024-09-01:
another related ticket: https://github.com/victronenergy/venus/issues/1059

ReinvdZee, 2024-12-10:
I changed the implementation such that a VRM device instance is requested dynamically and tied to the CAN gateway. The instance is stored in the setting:

`com.victronenergy/settings/Setting/Devices/<gateway>/ClassAndVrmInstance`

So each gateway gets its own instance, allowing multiple BMSes to be connected to one GX, but each over its own CAN interface. 

The battery monitor and selected BMS for DVCC will take the one with the lowest deviceinstance when set to automatic selection.

ReinvdZee, 2024-12-12:
Update: 

We're not going to support this now. This is only useful in a handful of edge (DIY) cases. Besides, it compilcates the currently stable working systems because a BMS plugged into a different CAN port will get a different VRM ID and may also need the DVCC settings redone.

closed 2025-01-02 by ReinvdZee