victronenergy/venus #1169: AC Loads calculation wrong when feeding to grid

mhin76, 2023-10-26:
I have installed an EM540 as grid meter on my venos OS v3.11 on Raspberry Pi. On Inverter I am using Hoymiles with a plugin that converts it to a venus readable inverter. I noticed (per phase) that as soon as a phase is feeding power to the grid the AC Load is wrongly calculated, it just shows the same amount as the PV Inverter. I see that others have the same issue (https://community.victronenergy.com/questions/221827/venusos-ac-loads-calculation.html)
This leads to totally wrong number also on my VRM history and shows always 100% solar consumption.
Could you check the bug?
[image]

DerJungeHerr, 2023-10-27:
Same Issue here! We need a fix soon 😔

mpvader, 2023-10-27:
sorry but thats just a not supported configuration. Or is there even one Victron device in your system?

mhin76, 2023-10-30:
apparently not. I was planning to create an ESS with a MultiPlus II. I know that hoymiles inverter are not supported and only show on my venus OS via dbus-opendtu. Apparently that integration is not sufficient to fully simulate a suported inverter.

mhin76, 2023-10-30:
Important info from the dbus_opendtu creator: in case you use hoymiles to visualize the inverter figures: that is only supported up to version 3 (version <3) of venus OS. apparently once connected with a MultiPlus it should also work with version 3 and above. will close this issue (as it is not a bug).

closed 2023-10-30 by mhin76