victronenergy/venus #1171: Generator start/stop: handle generators with their own stop delay

mpvader, 2023-11-06:
see here:

https://community.victronenergy.com/questions/240918/need-multiplus-to-ignore-generator-ac-during-cool.html

Sounds like making the 15 seconds configurable is a good and simple solution. (?).

mpvader, 2023-11-06:
Ps. can that user also work around this by disabling the cool down period?

kwindrem, 2023-11-06:
No, disabling cool-down won't help. The generator will still generate power after the GX device opens the relay, so the Multi will then pull power from the generator as it cools down.

The generator should shut down it's AC output during cooldown but apparently this is not the case.

vejpasop, 2023-12-01:
Can I hijack and reword this issue? I think instead of a configurable delay the code that handles the  "after cool down" should monitor if the AC on the input is actually gone. Only when gone the Multi should stop ignoring the AC input. (There should be a timeout around this check But this timeout might as well be a minute) 

In the current implementation the Multi stops ignoring the AC input 15 seconds after issuing the generator to stop.

kwindrem, 2023-12-01:
> Can I hijack and reword this issue? I think instead of a configurable delay the code that handles the "after cool down" should monitor if the AC on the input is actually gone. Only when gone the Multi should stop ignoring the AC input. (There should be a timeout around this check But this timeout might as well be a minute)
> 
> In the current implementation the Multi stops ignoring the AC input 15 seconds after issuing the generator to stop.
> I think this could cause undesirable results, especially in mobile applications that utilize an external ATS such as with the multiplus. If shore power is available and the generator is shut down there would still be power available at the ac input.

Yes, this would cause the post-cooldown period to run to it's limit if the ATS switches back to grid power.

My ExtTansferSwitch package changes the AC input type to/from generator and checks this for warm-up/cool-down and post cooldown so using the AC input would work, but NOT with the stock Victron code.

smith7977, 2024-01-19:
Hi folks – might be a different topic, but I'm learning that the Cerbo does not apply the genny warm-up delay setting when you do a manual start.  That's too bad, because the genny still needs warming up before taking on maximum load.  I often find it economical to run the genny for an hour or so at night to top up the the batteries when they're just low enough to need backup before dawn – but I'd rather not let the auto-start trigger itself and run the genny for six hours just before dawn.  Also, I like to exercise the genny once of week, whether the batteries need top-up or not.  Is someone working on adding this logical feature to the firmware?  Thanks

mpvader, 2024-01-20:
Hi @smith7977 ,

> I'm learning that the Cerbo does not apply the genny warm-up delay setting when you do a manual start.

we'll look into that, starting with seeing if its reproducible. Warm-up and cooldown should work, also on manual starts.

> Also, I like to exercise the genny once of week, whether the batteries need top-up or not.

No, nobody is working on that. Its an idea that came along now and then but was never really finished. Its on a list.

kwindrem, 2024-01-20:
Warmup and cooldown work for me with manual start. You do need a recent version of firmware on the Multi for ignore AC from the GX device to work but that would not be different for manual vs auto run.

Can't you use the periodic run to exercise the generator?

mpvader, 2024-01-20:
I stand corrected on a few points! Thank you @kwindrem .

@smith7977 , to be precise about the requirements for warm-up and cool-down:

1. GX Device must run Venus OS v3.00 or newer (please install v3.13, latest official release).
2. The MK3 in the GX Device must be updated, see "Manual step for MK3 update" chapter [in this blog](https://www.victronenergy.com/blog/2023/05/31/venus-os-v3-00-generator-start-stop-improvements-ess-peak-shaving/).
3. Inverter/charger firmware must be updated to a version equal to 502 or later.

And to make sure its clear: @smith7977 , this question by kwindrem is for you to answer:

> Can't you use the periodic run to exercise the generator?

smith7977, 2024-01-22:
Thanks all.  Here's what I have (newly installed in December):
Cerbo GX    Firmware version v2.91
Multi RS Smart 48V/6000VA/100A    Firmware version v1.15

It seems the Cerbo firmware is older than the 3.13 you recommend.  For the inverter firmware, the version I have seems unrelated to the 502 you recommend.

Good suggestion re using periodic run to exercise the genny – I'll look into that.  Thanks

mpvader, 2024-01-22:
Hi @smith7977 , I didn't expect you to use a Multi RS. With that inverter/charger it is a bit different then what I wrote before, sorry. Here it is:

All GX generator start/stop features work, ie automatic start/stop, quiet times, battery conditions, periodic runs.

Except for:

* Warm-up and cooldown, because for that to work, the GX needs to be able to tell the inverter/charger to not connect to the grid. That feature exists for the VE.Bus inverter/charger, but not for the Multi RS yet.
* Conditions related to the GX device, being AC load, Inverter high temperature and Inverter overload.

ReinvdZee, 2024-01-23:
> Hi @smith7977 , I didn't expect you to use a Multi RS. With that inverter/charger it is a bit different then what I wrote before, sorry. Here it is:
All GX generator start/stop features work, ie automatic start/stop, quiet times, battery conditions, periodic runs.
Except for:
> - Warm-up and cooldown, because for that to work, the GX needs to be able to tell the inverter/charger to not connect to the grid. That feature exists for the VE.Bus inverter/charger, but not for the Multi RS yet.

@mpvader, so trying to reproduce the issue with warm-up and cooldown is not necessary anymore?

ReinvdZee, 2024-01-25:
@jepefe , (Matthijs typing here), Rein will implement the initial plan (make the now hardcoded 15 seconds cooldown period configurable), and then ask you to review.

smith7977, 2024-01-28:
I figured out a workaround for the missing warm-up & cool-down on manual genny start: my system has a breaker in the line from genny to inverter.  So I can just click it off when starting up the genny, then click it on when the genny's good & warm, then click it off when desired charge or genny exercise is accomplished, and finally shut down the genny after a few minutes of cool-down.  That'll get me by while you guys look at software solutions.  Cheers

ReinvdZee, 2024-01-30:
> @jepefe , (Matthijs typing here), Rein will implement the initial plan (make the now hardcoded 15 seconds cooldown period configurable), and then ask you to review.

@jepefe, I've implemented and tested this feature. Can you take a look at [gui](https://github.com/victronenergy/gui/tree/rein/generator_postcooldowntime) and [dbus_generator](https://github.com/victronenergy/dbus_generator/tree/rein/generator_postcooldowntime)?

jepefe, 2024-01-30:
Reviewed, seems good to me.

mpvader, 2024-01-31:
@kwindrem : how about we call this setting `Generator stop time` ?

Now its `Post cool-down time` but sounds quite software-programmer-speak

ReinvdZee, 2024-01-31:
@kwindrem, @mpvader, Screenshot of the new setting:

[image]

kwindrem, 2024-01-31:
I agree, not the greatest name. It only says when the period occurs, not what it's for.

This is the time AFTER the GX device tells the generator to stop and before the AC input is again accepted. It is for cases where the generator does its own cool-down period and continues to produce power even though the system should be switching away from the generator.

Maybe "Ignore AC input after stop for ..."

This probably needs to be explained in the manual with a chart showing what happens when.

ReinvdZee, 2024-02-01:
@kwindrem, We changed it to `Generator stop time` and changed the menu structure a bit, see [this](https://victrondevelopment.slack.com/archives/C04RYU0UMDE/p1706687485368439) discussion on Slack.