victronenergy/venus #1175: hub4control: Make pv inverter offset configurable

izak, 2023-11-15:
## Background

In a system with both a Multi/Quattro and PV inverters, and where grid feed-in is limited, there is a possibility that the PV-inverter and the Multi competes to reach the ESS grid setpoint.

Once the setpoint is reached, a state of equilibrium results where the PV-inverter can not ramp up any further. If this state happens in a place where the Multi carries more of the load, or doesn't charge at full speed, you end up in a deadlock where PV is not fully utilised.

For this reason, hub4control has for years had an offset, where it will raise the PV-inverter setpoint by some amount whenever there is a PV-inverter that is in running state, so that the PV-inverter will always aim slightly lower than the Multi. This creates a tug-of-war (touwtrekkerij) around the setpoint, and favours the PV-inverter.

This mechanism was later enhanced, and instead of using a hardcoded offset, it was made to be 2% of the PV-inverter capacity. That way it scales with the PV-inverter size.

The offset cannot be made too large. If it is made too large, the Multi will have a tendency to import too much from the grid during the day.

## Problem

We STILL have sites where the offset just isn't large enough, and a non-optimal equilibrium arises at times. As stated before, a larger offset will cause more grid import, and a smaller offset creates more chance of non-optimal equilibrium.

## Suggested solution

Make the offset configurable. I'm generally not in favour of having too many configurable options, but this bug doesn't lend itself to many more options.

## TODO

- [ ] Hub4control modification
- [ ] GUI
- [ ] Log to VRM

mpvader, 2024-01-28:
What is the status of this Izak? On the todo it says this:

----

- Fix problem where batteries are not charged at full speed, while there is ample PV from a PV-inverter
  - This is done by adding a new setting for increasing the PV-inverter offset.
  - This offset is used to handicap the Multi, to ensure that the PV-inverter has priority. If the offset is too small, it can get locked up at a non-optimal point.
  - This is an addition to the improvement made in Venus 2.60, where this offset is always least 2% of the total PV-inverter capacity.
  - In most systems it is not necessary to change this. It is added only because other options have been exhausted.
  - https://github.com/victronenergy/venus/issues/1175
  
  ----
  
Was anything actually changed for users? or was just a new setting introduced, but not to the gui, so nothing fixed or changed for users yet?

In case that last is what the situation is: lets leave it at that then and finalise this in v3.30

mpvader, 2024-01-28:
(assigning it to v3.20 since I need to get the change log correct)

mpvader, 2024-10-11:
Lets discuss if we do this or not and but then close it