victronenergy/venus #1177: Fronius: Improve detection of 3-phase inverters using SolarApi

izak, 2023-11-16:
At the moment this relies on detecting the Device Type from a list provided by Fronius.

It already causes problems for the GEN24 and the Tauro, which has the same DT, but the GEN24 isn't 3-phase while the Tauro is. SolarAPI document says we should not rely on DT for the GEN24 and Tauro.

Further, every time Fronius brings out a new model, we will fall back to detecting it as a single phase unknown model.

Therefore, attempt to detect phase config a bit better.

Possible solution Idea: Calling `/solar_api/v1/GetInverterRealtimeData.cgi?Scope=Device&DataCollection=3PInverterData&DeviceId=1` returns an empty dictionary on single phase inverters.

The SolarAPI document is [here][1]. It contains an incomplete list of `DT` values, going backwards from 254 and down to 67 in this document, though obviously many models are not on this list and causing incorrect sunspec operation in the field.

[1]: https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42,0410,2012.pdf

Wholistic, 2023-11-22:
I am also seeing an uptick in complaints about this - would be nice to have resolved by 3.2 in line with the Sunspec improvements 

https://community.victronenergy.com/questions/241155/displaying-fronius-on-ccgx.html

izak, 2023-11-22:
The thing about this whole thing: There is an easy solution. Turn on Sunspec on the inverter. It is even documented. This is a problem because people don't read the documentation.

There are however a small hand full of cases where people cannot or will not use Sunspec, and unfortunately we did also create a situation where Fronius inverters usually work "out of the box", so fixing this is unavoidable :-)

closed 2023-11-30 by jhofstee