victronenergy/venus #1180: Vebus: Energy counter readout disabled on systems with many VE.Bus errors

vejpasop, 2023-11-28:
Potential bug since February 2015

We had a support case where occasionally the VE.Bus energy counters got stuck. Resetting Venus would resolve the issue.
After investigation,the energy counter readouts seem to get stuck around the time many Ve.Bus network errors are present on the bus.

mk2-dbus contains code to disable energy counter readout when the Multi responds that it does not support this feature or when slave Multi's report invalid energy counter data.
This feature was introduced as part of work around for Multi firmware version from before February 2015.

The most likely explanation is that due to the many communication errors the Multi responded with a command not supported which disabled further readout in mk2-dbus.

For that specific site. The Ve.bus network errors should be resolved. 

Todo:

- [ ] mk2-dbus: review code and make it retry instead of disabling readout.

mpvader, 2024-02-05:
@vejpasop can this be closed / is this fixed?

vejpasop, 2024-02-05:
This is fixed and can be closed

mpvader, 2024-02-05:
Included in v3.20-beta series.

(not in v3.14 !)