victronenergy/venus #1185: Victron rs485 usb dongle not recognized anymore

legraml, 2023-12-14:
I have two identical originial VE rs485 usb dongles, both labeled "USB-RS485-WE" on ttyUSB0 and ttyUSB1, ttyUSB0 is connected to EM24, ttyUSB1 is connected to ET112. They were both detected and running until some days ago when I made an upgrade to Cerbo GX latest firmware. Since then the dongle on ttypUSB0 is not detected anymore. When I plug it in, logs show:

Dec 14 11:44:58 einstein user.info kernel: [  386.958913] usb 4-1: new full-speed USB device number 4 using ohci-platform
Dec 14 11:44:59 einstein user.info kernel: [  387.245907] usb 4-1: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
Dec 14 11:44:59 einstein user.info kernel: [  387.254195] usb 4-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Dec 14 11:44:59 einstein user.info kernel: [  387.261429] usb 4-1: Product: USB <-> Serial
Dec 14 11:44:59 einstein user.info kernel: [  387.265740] usb 4-1: Manufacturer: FTDI
Dec 14 11:44:59 einstein user.info kernel: [  387.273241] ftdi_sio 4-1:1.0: FTDI USB Serial Device converter detected
Dec 14 11:44:59 einstein user.info kernel: [  387.280192] usb 4-1: Detected FT232RL
Dec 14 11:44:59 einstein user.info kernel: [  387.289525] usb 4-1: FTDI USB Serial Device converter now attached to ttyUSB0

product is not FT232R USB UART but USB <-> Serial
serial number is not shown.

When I plug this dongle into my PC and fire up FT_Prog, the dongle shows serial number and also the correct product id.

Compared to the dongle on ttyUSB1, they are idential besides having different serial numbers.

I really need help!

legraml, 2023-12-14:
Status update:

I could solve the problem by renaming the product id from "FT232R USB UART" to  "USB-RS485_Cable" by using FT-Programmer.
It was detected and started working immediately after plugging into the Cerbo GX.

Strange enough, the other dongle had exactly the same settings and still keeps running like a charm. So I don't really understand the problem.

mpvader, 2023-12-14:
Hey @legraml thank you for reporting and that is really strange. I'll remember it and for now consider it a one off / case closed.

closed 2023-12-14 by mpvader

legraml, 2024-01-25:
Unfortunately, this issue just appeared again after updating to the latest firmware for Cerbo GX. This time "USB-RS485_Cable" is not detected properly and the product type is listed again as: USB <-> Serial and not FT232R USB UART.

MasterCATZ, 2026-01-07:
I just want to chime in I had the same issue found this post and changing the device name helped

renamed 
FTDI USB-RS485-WE
to
FTDI FT232R USB UART

any possibility of adding more device names to the list ? 

```
root@raspberrypi2:~# dmesg | grep FTDI
[    4.611712] usb 1-1.2: Manufacturer: FTDI
[    4.881497] usb 1-1.1.2: Manufacturer: FTDI
[    5.130680] usb 1-1.3: Manufacturer: FTDI
[    5.547024] usbserial: USB Serial support registered for FTDI USB Serial Device
[    5.558410] ftdi_sio 1-1.2:1.0: FTDI USB Serial Device converter detected
[    5.586363] usb 1-1.2: FTDI USB Serial Device converter now attached to ttyUSB0
[    5.600727] ftdi_sio 1-1.1.2:1.0: FTDI USB Serial Device converter detected
[    5.622004] usb 1-1.1.2: FTDI USB Serial Device converter now attached to ttyUSB1
[    5.630678] ftdi_sio 1-1.3:1.0: FTDI USB Serial Device converter detected
[    5.655657] usb 1-1.3: FTDI USB Serial Device converter now attached to ttyUSB2
[    5.665662] ftdi_sio 1-1.1.3:1.0: FTDI USB Serial Device converter detected
[    5.783937] usb 1-1.1.3: FTDI USB Serial Device converter now attached to ttyUSB3
[ 1058.765149] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[ 1063.840464] usb 1-1.1.2: Manufacturer: FTDI
[ 1063.856922] ftdi_sio 1-1.1.2:1.0: FTDI USB Serial Device converter detected
[ 1063.871199] usb 1-1.1.2: FTDI USB Serial Device converter now attached to ttyUSB1
[ 1571.276284] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1
[ 1576.862022] usb 1-1.1.2: Manufacturer: FTDI
[ 1576.882906] ftdi_sio 1-1.1.2:1.0: FTDI USB Serial Device converter detected
[ 1576.900514] usb 1-1.1.2: FTDI USB Serial Device converter now attached to ttyUSB1
root@raspberrypi2:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.110 dwc_otg_hcd DWC OTG Controller
Bus 001 Device 002: ID 0424:2514  
Bus 001 Device 003: ID 0424:2514  
Bus 001 Device 004: ID 0403:6001 FTDI FT232R USB UART
Bus 001 Device 006: ID 0403:6001 FTDI FT232R USB UART
Bus 001 Device 007: ID 0403:6015 VictronEnergy MK3-USB Interface
Bus 001 Device 008: ID 0424:7800  
Bus 001 Device 010: ID 0403:6001 FTDI USB-RS485-WE
root@raspberrypi2:~# lsbusb
-sh: lsbusb: command not found
root@raspberrypi2:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.110 dwc_otg_hcd DWC OTG Controller
Bus 001 Device 002: ID 0424:2514  
Bus 001 Device 003: ID 0424:2514  
Bus 001 Device 004: ID 0403:6001 FTDI FT232R USB UART
Bus 001 Device 006: ID 0403:6001 FTDI FT232R USB UART
Bus 001 Device 007: ID 0403:6015 VictronEnergy MK3-USB Interface
Bus 001 Device 008: ID 0424:7800  
Bus 001 Device 010: ID 0403:6001 FTDI USB-RS485-WE
root@raspberrypi2:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.110 dwc_otg_hcd DWC OTG Controller
Bus 001 Device 002: ID 0424:2514  
Bus 001 Device 003: ID 0424:2514  
Bus 001 Device 004: ID 0403:6001 FTDI FT232R USB UART
Bus 001 Device 006: ID 0403:6001 FTDI FT232R USB UART
Bus 001 Device 007: ID 0403:6015 VictronEnergy MK3-USB Interface
Bus 001 Device 008: ID 0424:7800  
root@raspberrypi2:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.110 dwc_otg_hcd DWC OTG Controller
Bus 001 Device 002: ID 0424:2514  
Bus 001 Device 003: ID 0424:2514  
Bus 001 Device 004: ID 0403:6001 FTDI FT232R USB UART
Bus 001 Device 007: ID 0403:6015 VictronEnergy MK3-USB Interface
Bus 001 Device 008: ID 0424:7800  
root@raspberrypi2:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.110 dwc_otg_hcd DWC OTG Controller
Bus 001 Device 002: ID 0424:2514  
Bus 001 Device 003: ID 0424:2514  
Bus 001 Device 004: ID 0403:6001 FTDI FT232R USB UART
Bus 001 Device 007: ID 0403:6015 VictronEnergy MK3-USB Interface
Bus 001 Device 008: ID 0424:7800  
Bus 001 Device 011: ID 0403:6001 FTDI FT232R USB UART
Bus 001 Device 012: ID 0403:6001 FTDI USB-RS485-WE
root@raspberrypi2:~# Read from remote host 192.168.2.21: Connection reset by peer
Connection to 192.168.2.21 closed.
client_loop: send disconnect: Broken pipe
(base) aio@aio:~$ ssh root@192.168.2.21
X11 forwarding request failed on channel 0
Last login: Wed Jan  7 03:30:27 2026 from 192.168.2.47
root@raspberrypi2:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.110 dwc_otg_hcd DWC OTG Controller
Bus 001 Device 002: ID 0424:2514  
Bus 001 Device 003: ID 0424:2514  
Bus 001 Device 004: ID 0403:6001 FTDI FT232R USB UART
Bus 001 Device 007: ID 0403:6015 VictronEnergy MK3-USB Interface
Bus 001 Device 008: ID 0424:7800  
Bus 001 Device 011: ID 0403:6001 FTDI FT232R USB UART
Bus 001 Device 013: ID 0403:6001 FTDI FT232R USB UART
root@raspberrypi2:~# Connection to 192.168.2.21 closed by remote host.
Connection to 192.168.2.21 closed.
```