victronenergy/venus #1188: Not able to activate SHUTDOWN MONITOR

Brave7W, 2023-12-16:
I did the update from version 3.0 to 3.13. PI did not start after that, I did a reboot onto version 3 again. The "shutdown monitor" function was lost. I looked for it in the active packages. Its there as version 2.7 and under the installed menu it says "file set not found 3.0".

Will appreciate any help.

BW

kwindrem, 2023-12-16:
I have fixed ShutdownMonitor.

For future reference, this GitHub issues area is for developers only.

You need to post general issues to the Victron Community Forum or if specific to a third-party modification, to the issues section of that modification on GitHub.

Brave7W, 2023-12-16:
Thanks. New to the site, will find my way to the general section.

Cheers

closed 2024-01-09 by jhofstee