victronenergy/venus #1189: Move GUI alarms handling to Venus platform service

jepefe, 2023-12-18:
To ensure system-wide availability of alarms, rather than limiting them to the GUI, a notifications center has been implemented in the Venus platform. This implementation is based on the existing GUI notifications center, incorporating inherited code and logic from that it.

This implementation handles both the alarm relay and the buzzer.

venus-platform: 
https://github.com/victronenergy/venus-platform/tree/jperez/feat_platform_alarms

veutil:
https://github.com/victronenergy/veutil/tree/jperez/feat_platform_alarms

gui: (for reference)
https://github.com/victronenergy/gui/tree/jperez/WIP_platform_alarms/qml

```
Notifications/AcknowledgeAll			<- Acknowledge all the alarms
Notifications/Alarm				<- Alarm should/is triggered, buzzer is activated and the GUI will automatically switch the notifications page. 
Notifications/Alert				<- There are active notifications, but alarms are knowledged or there are only warnings. 
Notifications/NumberOfActiveNotifications	<- Number of active notifications
Notifications/NumberOfNotifications 		<- Number of total notifications
Notifications/[0-19]/Acknowledged		<- Acknowledged status
Notifications/[0-19]/Active			<- Condition that triggered the alarm is active or Notifications
Notifications/[0-19]/AlarmValue			<- Value of the alarm item/trigger
Notifications/[0-19]/DateTime			<- Date and time of when the notification was generated
Notifications/[0-19]/Description		<- Description of the alarm, translated to the language selected in the GUI
Notifications/[0-19]/Service			<- Service that the alarm is related to
Notifications/[0-19]/Trigger			<- Item of the service that contains the alarm status
Notifications/[0-19]/Type			<- Type of notification 0: Warning, 1: Alarm, 2: Notification
Notifications/[0-19]/Value			<- Value that is the cause of the alarm at the moment the alarm was triggered, for example, the temperature on a high temperature alarm
```

There is a second phase, that will include a centralised document where error codes and is description will be defined. The scope of this first implementation is to replace the functionality that it is currently in the gui.

jepefe, 2023-12-18:
@jhofstee Please, take a look and merge if you consider it good.

jhofstee, 2024-03-29:
yes

closed 2024-03-29 by jhofstee