victronenergy/venus #1190: Make installer image instructions

mpvader, 2023-12-20:
### TODO

- [ ] Make these instructions for Ekrano GX, Cerbo GX and Color Control GX and Venus GX. We have separate manuals for each of them, no need to make one generic instruction. Stefanie will be adding it to the manuals, by splitting up the Factory reset chapter into two settings, so Chapter 18.1: Reset to factory defaults. Chapter 18.2: Venus OS reinstall. With instruction: "Use this when the procedure at 18.1 doesn't work"
- [ ] Make a venus-data-restore-wifi-Cerbo-prior-to-HQ2026.tar.gz

When working on this issue, make sure to read these two links:
- https://community.victronenergy.com/questions/204255/cerbo-gx-bricked-how-to-recover.html
- https://victrondevelopment.slack.com/archives/C7C1ZU6UD/p1683464175843239

### Concept text for in that chapter 18.2:

WARNINGS:

- before executing this procedure, first try the common factory reset procedure as in the section above above.

- only perform this procedure as a last resort: to fix a bricked device. A device that boots fine, but has some strange behaviour in some feature, will not benefit from executing this procedure.

- all data on the data partition will be wiped by this procedure. Meaning all settings, and so forth.

- this procedure does, unlike to common “reset to factory defaults instruction”, not rely on a properly booting device.

- when executed on a Cerbo GX prior to HQ2026, the wifi access point as well as some other (non crucial) feature won’t work anymore; since some factory installed data is missing. Per HQ2026 that data is stored in a more secure place (eeprom instead of data partition).

- On the VRM Portal you’ll have to reset the Device token. No new data will be accepted by the portal until that is done.

PROCEDURE:

1. Download the installer image: http://updates.victronenergy.com/feeds/venus/release/images/einstein/venus-install-sdcard-einstein.img.zip
2. Write that to a sdcard using the Balena Etcher application.
3. When using a GX Touch, wait for the process to be finished:

[image]

If there is no GX Touch available, then wait for 2 minutes.

3. Remove the SDcard and power cycle the GX device.

mpvader, 2023-12-20:
For a CCGX, this is our instruction, and rather than having that as an "On request" instruction sent by the repairs department I'd much rather have it in the manual. Maybe also referred to in the Pre-RMA doc btw.

I would like to ask you to try an update when blue/white screen is shown. (This also applies for Victron screen).
If you want to try fixing a unit stuck on a screen using the SD card try the method described here:
https://www.victronenergy.com/live/ccgx:firmware_upgrade_to_v2 
If this doesn’t work let us know and we will replace the CCGX.

ReinvdZee, 2024-01-10:
### Installer image instructions per machine

### CCGX
_note: This procedure requires a functional U-boot on NAND flash._

1. Download the upgrade image here: https://updates.victronenergy.com/feeds/venus/release/images/ccgx/  (venus-upgrade-image-ccgx-*.sdcard.zip)
2. Format your micro-SD card or USB flash drive as a FAT32 filesystem. 
3. Extract the contents of the archive to the micro-SD or USB drive. Do not use a flashing application (like Balena Etcher) for this. 
4. Insert the micro-SD card or USB flash drive in the CCGX device. 
5. Press and hold the left button of the five-button central pad. 
6. Power up the CCGX device, the update process will now start, progress is reported on the LCD-display.
7. Release the button and wait for the process to finish. The user is informed to remove the installer medium (micro-SD card or USB flash drive) and to powercycle the system. 
8. Remove the micro-SD card or USB drive and powercycle the CCGX device. 

### Venus GX

1. Download the installer image here: https://updates.victronenergy.com/feeds/venus/release/images/beaglebone/ (venus-install-sdcard-beaglebone-*.img.zip)
2. Flash image to a micro-SD card using the Balena Etcher application (https://etcher.balena.io/). The Etcher application automatically unzips the archive.
3. Insert the micro-SD card into the Venus GX device.
4. Press and hold the switch to the right of the green connector. 
5. Power up the device.
6. Release the switch when the LED (left of the green connector) starts flashing red.
7. Wait until the LED is fully green, The update process is now finished.
8. Remove the micro-SD card and powercycle the device. 

### Cerbo GX

1. Dowload the installer image here: https://updates.victronenergy.com/feeds/venus/release/images/einstein/ (venus-install-sdcard-einstein-*.img.zip)
2. Flash image to a micro-SD card using the Balena Etcher application (https://etcher.balena.io/). The Etcher application automatically unzips the archive.
3. Insert the micro-SD card into the cerbo GX device.
4. Power up the device.
5. Wait for the installation process to finish. When connected to GX touch, the user is informed to remove the installer medium (micro-SD card) and to powercycle the system. If not connected to GX touch, wait for 2 minutes for the installation process to finish.
6. Remove the micro-SD card and powercycle the device. 

### Ekrano GX

1. Dowload the installer image here: https://updates.victronenergy.com/feeds/venus/release/images/ekrano/ (venus-install-sdcard-ekrano-*.img.zip)
2. Flash image to a micro-SD card using the Balena Etcher application (https://etcher.balena.io/). The Etcher application automatically unzips the archive.
3. Insert the micro-SD card into the ekrano GX device.
4. Power up the device.
5. Wait for the installation process to finish. The user is informed to remove the installer medium (micro-SD card) and to powercycle the system.
6. Remove the micro-SD card and powercycle the device.

mpvader, 2024-05-15:
Meanwhile, the chapters have been added to the manuals.

But what we missed is the EasySolar-IIGX as well as the MultiPlus-II GX; @ReinvdZee can you make & test an instruction for that as well? See mail for the trigger.

ReinvdZee, 2024-05-21:
@mpvader, I've updated the original comment with instructions for the Multiplus-II GX and Easysolar-II GX.

ReinvdZee, 2024-06-04:
Instructions on how to access the SD card slots of the Multiplus-II GX and EasySolar-II GX.

**Multiplus-II GX**
1. Remove all connections from the bottom side of the unit.
2. Unscrew 13 screws to remove the cover. 4 on each side, 3 on the bottom side (two larger and 1 smaller one) and 2 on the backside of the unit. 
[image]
[image]
3. Remove the cover to find the panel with the GX card. Mind the flatcable connected to the display on the cover.
[image]
4. Lift the panel holding the GX card. Mind the cabling to the GX card. With the cables attached, the panel can be flipped over to access the GX card. The nanopi board is the blue rectangular board on the GX card, indicated with the red rectangle.
[image]
5. The SD card slot is found on the bottom side of the nanopi board:
[image]

**Easysolar-II GX**
1. Unscrew the cover of the unit. 3 screws on each side and 2 on the bottom side.
2. Remove the cover to find the panel with the GX card. Mind the flatcable connected to the display on the cover. The GX card is indicated with the red rectangle. 
[image]
3. The SD card slot is found on the bottom side of the nanopi board. The SD card slot should be accessible without removing the GX card from the unit. 
[image]

alnavasa, 2025-02-27:
It is not clear what will happen to a Cerbo GX previos to HQ2026
`when executed on a Cerbo GX prior to HQ2026, the wifi access point as well as some other (non crucial) feature won’t work anymore; since some factory installed data is missing. Per HQ2026 that data is stored in a more secure place (eeprom instead of data partition).`
What features wont work? Is there anyway to solve this and do a reinstall on the pre HQ2026 models, without loosing crucial and non crucial features?

ReinvdZee, 2026-10-12:
Note on the Cerbo GX Mk3 (Also added to my earlier comment above):

In order to boot from the SD card, the user needs to press and hold the network button (right above the power connector).
The Mk3 can be identified with its article code: BPP900450130 or BPP900451130

ReinvdZee, 2026-10-12:
Also added the instructions for the Nucleo GX.