victronenergy/venus #1191: resize2fs.sh fails on CCGX

kwindrem, 2024-01-03:
This script is the recommended way to prepare a GX device for modifications to the root partition. The script can not run on CCGX because of the following error: Rootfs is not ext4, but ubifs

I'm not sure this is a bug and it appears the file system is probably as large as it can be so a resize would not be of any help. But at least a procedure change is needed to prepare a CCGX for modifications.

jhofstee, 2024-01-04:
If it really says error, then it is a bug. It is intended as simply informing the user why the fs is not resized. It is not ext4 in this case and doesn't need resizing.

kwindrem, 2024-01-04:
The message is: "Rootfs is not ext4, but ubifs" so no, not an error.

However, resize2fs also remounts root as read-write and this is not happening if the ubifs format is detected. The script exits before the code to remount is executed.

The issue hinges around what is stated in the Venus OS: Root Access document. It recommends using resize2fs. No mention of remount-rw or CCGX issues. 

Fixing resize2fs to remount root even for ubifs format would simplify things for developers. But an update to the documentation would also be sufficient I think.

What I did in SetupHelper was to separate the remounting of the root as read-write using remount-rw.sh and resizing the root partition with resize2fs.sh. Resizing is skipped for CCGX.

jhofstee, 2024-01-09:
"However, resize2fs also remounts root as read-write and this is not happening if the ubifs format is detected."

Are you sure about that? Or is it simply the message that it won't be resized that made you thing so? At least I fail to reproduce it....

release version:
```
root@ccgx:~# cat /opt/victronenergy/version 
v3.13

root@ccgx:~# mount
ubi0:rootfs1 on / type ubifs (ro,relatime,assert=read-only,ubi=0,vol=0)

root@ccgx:~# /opt/victronenergy/swupdate-scripts/resize2fs.sh
making the rootfs writable...
Rootfs is not ext4, but ubifs. Exit.

root@ccgx:~# mount
ubi0:rootfs1 on / type ubifs (rw,relatime,assert=read-only,ubi=0,vol=0)

root@ccgx:~# cat /etc/fstab 
/dev/root / auto defaults 0 0
proc                 /proc                proc       defaults              0  0
```

testing:
```
v3.20~36
root@ccgx:~# mount
ubi0:rootfs2 on / type ubifs (ro,relatime,assert=read-only,ubi=0,vol=1)

root@ccgx:~# cat /etc/fstab 
/dev/root            /                    auto       ro              0  0

root@ccgx:~# /opt/victronenergy/swupdate-scripts/resize2fs.sh 
making the rootfs writable...
Rootfs is not ext4, but ubifs
modifying /etc/fstab

root@ccgx:~# mount 
ubi0:rootfs2 on / type ubifs (rw,relatime,assert=read-only,ubi=0,vol=1)

root@ccgx:~# cat /etc/fstab 
/dev/root / auto defaults 0 0
proc                 /proc                proc       defaults              0  0
```

kwindrem, 2024-01-09:
My original code based its behavior on the return code of resize2fs.sh. Since there are no explicit return statements, maybe that is indeterminate. If so, that would have caused my code to abort the install. My bad.

I have since modified my code and check the mount status rather than script return codes. So all is well.