victronenergy/venus #1197: Move settings creation from gui-v1 elsewhere

mpvader, 2024-01-10:
List comes from: https://github.com/victronenergy/gui/blob/master/src/ve_qitems_settings.cpp.

```
	add("Alarm/Audible", 1, 0, 0);
	add("Alarm/Vebus/HighDcCurrent", 1, 0, 2);                 -> move to mk2-dbus itself, so @vejpasop 
	add("Alarm/Vebus/HighDcRipple", 1, 0, 2);                  -> same
	add("Alarm/Vebus/HighDcVoltage", 1, 0, 2);                 -> same
	add("Alarm/Vebus/HighTemperature", 1, 0, 2);               -> same
	add("Alarm/Vebus/InverterOverload", 1, 0, 2);              -> same
	add("Alarm/Vebus/LowBattery", 1, 0, 2);                    -> same
	add("Alarm/Vebus/TemperatureSenseError", 1, 0, 2);         -> same
	add("Alarm/Vebus/VeBusError", 2, 0, 2);                    -> same
	add("Alarm/Vebus/VoltageSenseError", 1, 0, 2);             -> same
	add("Gps/Format", 0, 0, 0);                                -> veplatform  @ReinvdZee 
	add("Gps/SpeedUnit", "km/h");                              -> veplatform   @ReinvdZee 
	add("Gui/AutoBrightness", 1, 0, 1);                        -> veplatform    @ReinvdZee 
	add("Gui/Brightness", platform.getMaxBrightness(), 0, platform.getMaxBrightness());    -> veplatform @ReinvdZee 
	add("Gui/DefaultOverview", "");                            -> don't move; each gui should have its own path for this.
	add("Gui/DisplayOff", 600, 0, 0);                          -> veplatform    @ReinvdZee 
	add("Gui/Language", "en");                                 -> veplatform    @ReinvdZee 
	add("Gui/MobileOverview", 0, 0, 1);                        -> don't move
	add("Gui/StartWithMenuView", 0, 0, 1, true);               -> don't move
	add("Gui/TanksOverview", 0, 0, 1);                         -> don't move
	add("Gui/TouchEnabled", 1, 0, 1);                          -> veplatform @ReinvdZee 
	add("Sensors/OnPosition/ACIn1_L1", "");                    -> don't move (part of qwacs, which is deprecated)
	add("Sensors/OnPosition/ACIn1_L2", "");                    -> don't move (part of qwacs, which is deprecated)
	add("Sensors/OnPosition/ACIn1_L3", "");                    -> don't move (part of qwacs, which is deprecated)
	add("Sensors/OnPosition/ACIn2_L1", "");                    -> don't move (part of qwacs, which is deprecated)
	add("Sensors/OnPosition/ACIn2_L2", "");                    -> don't move (part of qwacs, which is deprecated)
	add("Sensors/OnPosition/ACIn2_L3", "");                    -> don't move (part of qwacs, which is deprecated)
	add("Sensors/OnPosition/ACOut_L1", "");                    -> don't move (part of qwacs, which is deprecated)
	add("Sensors/OnPosition/ACOut_L2", "");                    -> don't move (part of qwacs, which is deprecated)
	add("Sensors/OnPosition/ACOut_L3", "");                    -> don't move (part of qwacs, which is deprecated)
	add("System/AccessLevel", 1, 0, 3);                        -> veplatform    @ReinvdZee 
	add("System/AutoUpdate", 2, 0, 3);                         -> veplatform    @ReinvdZee 
	add("System/LogLevel", 2, 0, 0);                           -> veplatform    @ReinvdZee 
	add("System/ReleaseType", 0, 0, 3);                        -> veplatform    @ReinvdZee 
	add("System/TimeZone", "/UTC");                            -> veplatform    @ReinvdZee 
	add("System/Units/Temperature", "");                       -> veplatform    @ReinvdZee 
	add("System/VolumeUnit", 0, 0, 0);                         -> veplatform    @ReinvdZee 
	add("SystemSetup/SystemName", "");                         -> veplatform    @ReinvdZee 
#ifdef QML_TEST
	add("Gui/FreeTextTest", "test");                      -> don't move
	add("Gui/RangeTestHigh", 1.1, 0.0, 1.3);         -> don't move
	add("Gui/RangeTestLow", 0.0, -0.2, 1.0);         -> don't move
	add("Gui/OptionTest", 3, 0, 3);                       -> don't move
	add("Gui/TimeTest", 1, 0, 86400);                   -> don't move
	add("Gui/SpinBoxTest", 5, 4, 12);                       -> don't move

	add("Gui/Test/Items/0/Value", 5, 0, 12);                   -> don't move
	add("Gui/Test/Items/1/Value", 1, 0, 12);                   -> don't move
	add("Gui/Test/Items/2/Value", 3, 0, 12);                    -> don't move
```

mpvader, 2024-01-10:
I listed this for v3.20, but actually can be moved to v3.30 as well: as long as its done when releasing the new gui we're fine.

ReinvdZee, 2024-01-15:
I've added the required settings to venus-platform (veplatform). 
I assume these settings need to be removed from GUI-V1 as well, correct?

mpvader, 2024-01-15:
nice. and indeed they need to be removed from gui-v1

ReinvdZee, 2024-01-15:
Great, will do.

closed 2024-02-08 by jhofstee