victronenergy/venus #1201: v3.20: crashes when changing the time?

mpvader, 2024-01-24:
https://community.victronenergy.com/questions/249509/venus-os-v32034-available-for-testing.html?page=2&pageSize=50&sort=oldest

[image]

@ReinvdZee can you triage / see if you can reproduce?

ReinvdZee, 2024-01-25:
I tested this quickly on v3.20~37 and v3.20~40. On these versions, I am unable to reproduce the issue.

If someone points me to the location of the v3.20~35 release, then I can also test on this version.

mpvader, 2024-01-25:
Hi @ReinvdZee , we don't keep those. Only official release are archived. I've answered on Community saying that we can't reproduce, and asking him for his URL and to enable Remote Support.

Lets await that.

mpvader, 2024-02-02:
Received no report back from the user; closing.

closed 2024-02-02 by mpvader