victronenergy/venus #1202: Unfortunately, this issue just appeared again after updating to the latest firmware for Cerbo GX. This time "USB-RS485_Cable" is not detected properly and the product type is listed again as: USB <-> Serial and not FT232R USB UART.

legraml, 2024-01-25:
Unfortunately, this issue just appeared again after updating to the latest firmware for Cerbo GX. This time "USB-RS485_Cable" is not detected properly and the product type is listed again as: USB <-> Serial and not FT232R USB UART.

_Originally posted by @legraml in https://github.com/victronenergy/venus/issues/1185#issuecomment-1910125316_

Pearsonhenry2, 2024-04-21:
I have a very similar issue but I am using a waveshare 4 channel RS485 converter.
Surely a Cerbo can handle a non-victron RS485-USB converter (it does just fine with it for connecting to JK BMS)

[image]

thenebu, 2026-08-29:
Hit the same issue on a Cerbo GX after updating v3.75 -> v3.79 (kernel 6.12.90-venus-4). Two Carlo Gavazzi meters (EM340 grid + EM24, daisy-chained on one official Victron USB-RS485 cable) disappeared simultaneously.

Confirmed root cause: the adapter's USB descriptor now reports `ID_MODEL=USB__-__Serial` ("USB <-> Serial", no serial number) instead of `FT232R_USB_UART`. Since `/etc/udev/rules.d/serial-starter.rules` only tags the latter as `rs485:default`, the port silently fell back to the `default` alias (gps/vedirect only) and `dbus-cgwacs` was never started on it — no error anywhere, serial-starter just cycles through the wrong candidates on that tty forever.

Rather than reprogramming the FTDI EEPROM with FT-Prog (which is what worked for @legraml above but permanently rewrites the device and has to be redone per-cable), we worked around it by matching the fixed physical USB port path instead of the product string, via a udev rule dropped into `/data` so it survives future firmware updates:

`/data/conf/rs485-fix/install.sh`:
```bash
#!/bin/bash
set -e
mkdir -p /etc/udev/rules.d
cat > /etc/udev/rules.d/zz-rs485-em340-em24-fix.rules <<'RULE'
ACTION=="add", SUBSYSTEM=="tty", KERNELS=="4-1", ENV{VE_SERVICE}="rs485:default"
RULE
udevadm control --reload-rules
```
(`4-1` is this Cerbo's fixed physical USB port for that cable — find yours via `udevadm info -q all -n /dev/ttyUSBx` and look at `DEVPATH`.)

Registered in `/data/rc.local` so it reapplies on every boot:
```bash
bash /data/conf/rs485-fix/install.sh
```

After running it once and firing `udevadm trigger --action=add --subsystem-match=tty --sysname-match=ttyUSB0`, `dbus-cgwacs` immediately found both meters again — no reboot or cable replug needed.

Given this has now recurred across at least two separate firmware updates for multiple people, might be worth either widening the `serial-starter.rules` match (e.g. also match on `ID_VENDOR_ID=="0403" && ID_PRODUCT_ID=="6001"` generically, falling back to `rs485:default` for unbranded/blank-EEPROM FT232R clones) or shipping a way to pin `VE_SERVICE` by device path/serial through the GUI instead of requiring a hand-rolled udev rule.

thenebu, 2026-08-29:
Follow-up with a concrete root cause, in case it helps anyone else hitting this: I read the FTDI EEPROM directly (via libftdi's `ftdi_read_eeprom`/`ftdi_get_eeprom_buf`, vendor-specific EEPROM read requests, not the normal USB string-descriptor path) and got:

```
Checksum Error: 8868 c868
manufacturer=FTDI  product="FT232R USB UART"  serial="B00 4LCN"
```

So the EEPROM content itself is still intact and correct (`FT232R USB UART`, real serial number) — but its **checksum no longer validates**. The FT232R validates this checksum in-chip at every power-up/USB reset; when it fails, the chip's own enumeration apparently doesn't reliably serve the stored string descriptors anymore, and the host falls back to a generic `USB <-> Serial` with no serial number. This looks non-deterministic (timing-dependent at power-up), which matches the pattern in this thread: it can run fine for a long time and then flip after any reboot/power-cycle — a firmware update just happens to force that reboot.

So on my unit at least this isn't a driver/software regression at all, it's a genuinely degrading EEPROM (cable was ~2 years in continuous service). Might be worth adding a defensive workaround on the Venus side regardless (e.g. treat unbranded/blank FT232R descriptors as a `rs485` candidate too, per my udev suggestion above), since these cables apparently degrade this way over time and a firmware update's reboot is often what surfaces it.

mpvader, 2026-11-04:
Thanks @thenebu for finding the cause. For anyone hitting this: a VE USB RS485 cable whose EEPROM got corrupted reports "USB <-> Serial" instead of "FT232R USB UART", so the GX no longer tries the meter drivers on it. Replacing the cable fixes it as well.

We might fix this in Venus OS as well some day, to also accept cables with such failed eeprom. But that requires proper testing and such -> there are other prios at the moment.