victronenergy/venus #1204: v3.20: BatteryLife no longer works

mpvader, 2024-01-28:
https://community.victronenergy.com/questions/257253/venus-os-v32040-available-for-testing.html

[image]

RandMental, 2024-01-29:
ESS Issues - Loaded Beta 41 this morning.

Easy Soler 48V/5000W system that does not want to feed back into grid whatever the ESS and GX settings (GX V3.13 and V3.14 tried)
•	MultiPlus 48/5000/70-100		             V5.08   (2624)
•	Color Control GX                                          V3.13/14 
•	SmartSolar MPPT VE.Can 250/100 rev2        V3.14
•	Pylontech 2 x US3000C as master              ~V1.7  (new version)
 		     2 x US3000A	
 		     
This is in the SA market with “ Not complying with any Grid Code Standard”  plus LOM B (safe)  set. and UPS mode with ESS Assistant.

I can confirm that the ESS  "Keep batteries charged", also still don' t work correctly.  

In the Screen below the 48V DC coupled system is capable of 5kW,  yet the Pylontech US3000C  batteries stays on ~95% SOC  as previously set and tried with and without BatteryLife>
No Grid feed 
 
[image]

With the Grid available it still don’t feedback into the grid as expected

[image]

izak, 2024-01-30:
I think I know why. Auto-recharge appears to be broken.

RandMental, 2024-01-31:
Hi Izak, what would you need from us to assist solving the above?

izak, 2024-01-31:
@RandMental Your issue appears to have nothing to do with the actual bug reported above, so it is unlikely that you can help me with this one.

izak, 2024-01-31:
This issue is definitely present on Harold's test system, so I'm continuing to debug it there. Original customer that reported the issue is unresponsive.

izak, 2024-01-31:
Bug found. It affects sites without a managed battery, and with no maximum discharge power set.

In technical terms, doing math in C++ with a NaN (not a number) yields NaN answers and then nothing works down the line. For sites without a managed battery (and no ESS max discharge power set), the maximum discharge power is a NaN.

RandMental, 2024-02-01:
Dankie Izak, Our Installers willbe on-site this morning to check what is happening here

izak, 2024-02-01:
Bug fixed in hub4control v1.3.3, and on the TODO.