victronenergy/venus #1207: venus-platform & guis: make venus platform monitor RunningVersion and action it

mpvader, 2024-01-31:
In localsettings there is a setting `/Settings/Gui/RunningVersion`. Both gui-v1 and gui-v2 use that: the user can select which on screen display he wants to use.

Todo:
- [ ] Add code to venus-platform to monitor that setting, and make the running onscreen gui stop, or restart the start-gui service in case it changes. (@ReinvdZee )
- [ ] Finish and include this PR in gui-v2
- [ ] When including that new version of gui-v2, remove [this patch](https://github.com/victronenergy/meta-victronenergy-private/blob/master/recipes-ve/gui-v2/0001-add-an-UI-version-switch.patch) from meta-victronenergy-private.

Background:

We want users to be able to switch from gui-v1 to gui-v2 and back, for the On screen display.

At the moment its like this:
- In gui-v1, that setting is available both on screen and in remotely (remote console)
- In gui-v2, that setting is available only in the build that runs on the screen (see above patch).

Here is how we'll add and word it such in gui-v2, also when seen in a browser, that its clear that its about the onscreen display: https://github.com/victronenergy/gui-v2/commit/1f4b147b037a377575906a6d451c156bc8c45b1c.

ReinvdZee, 2024-01-31:
I tied `/Settings/Gui/RunningVersion` to the service `/service/start-gui`([see commit](https://github.com/victronenergy/venus-platform/commit/9fe09c3d7b51e4edbebfdcf7ac039eeba295ae8a)). The service restarts whenever `/Settings/Gui/RunningVersion` changes (allowed values are 1 or 2). 

I tested this by changing `/Settings/Gui/RunningVersion` directly and by using the buttons in the GUI. Both work fine.

A sidenote, when switching from `Beta UI` to `standard UI`, the standard UI is shown on the display briefly (less than a second) before restarting. I'm not quite sure why this happens and if it really is a problem.

kwindrem, 2024-01-31:
Note that other platforms, notably, Raspberry PI have local displays so it's not just Cerbo with GX Touch and Ekrano. (CCGX isn't an issue because it will always run gui-v1 locally.)

Not sure I'm reading the code correctly but is a reboot really necessary? If this is happening only when the value is not 1 or 2, does this get the system stuck in a constant reboot?

As a side note: many have been confused or complained about not being able to access the GUI via VRM remote console. My tests indicate no issues running gui-v1 headless and gui-v2 at the same time so maybe this is a stop-gap until the VRM remote console can access gui-v2.

mpvader, 2024-02-01:
Kevin, this is going to be fine, no need to worry, please just wait and see what is coming.

ReinvdZee, 2024-02-01:
@kwindrem, the system won't reboot, only the `start-gui` service will restart. Further, the service will only be restarted when `/Settings/Gui/RunningVersion` changes, and it can only be either 1 or 2.

mpvader, 2024-02-13:
hey @ReinvdZee , iirc this is all included now, so will be part of v3.30.

Right?

then it can be closed.