victronenergy/venus #1209: feature: possible support of Bluetooth Rotarex LPG gauge

mman, 2024-01-31:
Rotarex SRG manufactures LPG autofilling valves with measurement functionality that work great with German Alugas bottles.

https://rotarexsrg.com/product/dimes-camping-europe-488-012-2001
https://rotarexsrg.com/product/dimes-camping-us-487-013-2001

Their benefit is that they are mounted as part of the filling valve and level sensor battery is easily accessible compared to for example Mopeka or Truma LPG sensors that are underneath the bottle.

The valves advertise via Bluetooth LE and periodically send LPG level and sensor battery level as part of advertising payload.

I do not have official documentation, but by observing the system and its advertising data, I could find that byte 2 in advertising payload represents battery level from 0-100 and byte 3 indicates lpg level from 0-100.

Feel free to close if not appropriate, I am dropping it here since the implementation seems identical to current implementation of Mopeka sensors and may be of benefit to Victron users.

[image]

stenekholm, 2024-04-09:
Yea please!!

mman, 2024-05-17:
Got a confirmation from Rotarex engineer that the investigation above is correct. Should be fairly easy to implement by simply duplicating the existing Mopeka code and adjusting the advertising payload parser to extract battery and LPG level.

LAN-Opfer, 2025-04-06:
Has anyone managed to create a working code (e.g., C++ for EPS32) to read a Rotarex SRG? Unfortunately, I can't find the UUID DDA86642-B3C5-1DA1-C9BE-8D05C9DE9844 using "nrf connect" on Android.

mman, 2025-04-07:
@LAN-Opfer I have a working code in node red where I simply observe the Bluetooth advertisements and parse out the LPG level and batter level… works great…

Alex-DeIt, 2026-03-22:
I used two ways for obtaining the raw data: Using BLE Monitor I obtained 07 | ff | ff | ff | ec | 42 | 5f | 5f, , while using a esp32 bridge I obtained 07 | 43 | 5f | 5f I and did not recognize, that those data simply correspond (more or less) to the last 4 byte of the raw data I obtained with BLA monitor. So, pls. do not consider my question related to the 4 and 8 byte lenght.