victronenergy/venus #1215: Add network control to Venus-platform

ReinvdZee, 2024-02-07:
Gui-v2 requires the network settings to be exposed to dbus. see victronenergy/gui-v2#618.

Gui-v1 interacts with connman directly, gui-v2 will not, so the settings need to be exposed to dbus.

ReinvdZee, 2024-02-20:
Controlling ethernet using connman from `venus-platform` is now working properly. 

Next, WiFi must also be controlled from `venus-platform` and exposed to the dbus. Applications (gui) can then connect to and configure the network over dbus. 

**Dbus service per WiFi network**
An option would be to create a dbus service for every discovered network and delete the service when the network is out of range. I think this is the best solution but can lead to lots of services, again depending on the number of networks.

**Set of settings per WiFi network**
Another option is to create a set of settings (under `Settings/Network/...`, e.g.). However, this would lead to the creation of quite some dbus settings, depending on the number of WiFi networks discovered. In addition, this also requires dbus settings to be removable after creation when a network is out of range. I have not found this option in `VeQItemSettings`. Is this even supported? And is this compliant with the Victron DBUS api?

**Ethernet**
Since the ethernet connection is also represented by a service in connman, it makes sense to manage it the same way as the WiFi networks. 

Any thoughts? @jhofstee, @mpvader?

ReinvdZee, 2024-03-26:
Network control was added to venus-platform and gui, see https://github.com/victronenergy/gui-v2/issues/618.

jhofstee, 2024-05-23:
gui-v1 changes:
- When a networks is not connected to yet, the IP configuration is now shown as Unknown, instead of automatic.
- The unknown ip info used displayed as --, but changed to empty strings
- TypeError: Result of expression 'services["wifi"][networkName]' [undefined] is not an object get reported quite often.
- The order is a of the wifi networks seems to be [Name, connected] so the connected ones end up at the bottom instead of the top.
- the selected index isn't maintained (because of the array as model instead of a abstract model derived one)

venus-platform:
- it an report agent already registered. Don't know how to reproduce that at the moment
- state info seems outdated at times, e.g. disconnected while being connected etc. No idea how to reproduce that either.
- (or gui-v1) branch, when switching from automatic to manual, I ended up with a 3 digit dns server. Can't reproduce either.

dbus-tool
- dbus-y doesn't like spaces in wifi names. Not important but looks confusing.

I will include the venus-platform v1.40, so at least gui-v2 can start using it and the we
can check if mentioned is can be reproduced there as well or not.

I pushed my gui-v1 branch here https://github.com/victronenergy/gui/commits/jeroen/network_control/.
It now serializes the json code before sending. The bluetooth availability is now reported by venus-platform.

mpvader, 2024-06-14:
@jhofstee am I correct in seeing that:

1. all changes are included venus-platform
2. for gui-v1 we didn't include the changes for this
3. so far all works well, so can be released as is in v3.40 ?

jhofstee, 2024-06-21:
yes, there is room for improvement, like sorting the networks etc. But that is better then not being able to configure the network at all ;)

jhofstee, 2026-10-07:
The most basic things are still lacking here, like sorting by the Wifi with the best reception. (In gui-v2 that is, gui-v1 does that)