victronenergy/venus #1216: Add dbus path to venus-platform for switching between percentages and units on the circular gauges

ReinvdZee, 2024-02-07:
See victronenergy/gui-v2/#551. 

Some dbus path needs to be exposed by venus-platform for gui-v2 to use.

mpvader, 2024-02-12:
Hey @ReinvdZee to make this 100% clear: venus-platform doesn't need to expose this path, it just needs to make sure its created as a setting. or not?

ReinvdZee, 2024-02-12:
@mpvader, 

venus-platform will create a dbus path if it does not exist and assign it a default value. The gui can then use it as desired. 

Also see my comment on https://github.com/victronenergy/gui-v2/issues/551.

mpvader, 2024-02-12:
After discussing, this is already done, see here:

https://github.com/victronenergy/meta-victronenergy-private/blob/master/recipes-ve/gui-v2-webassembly/localsettings

closed 2024-02-12 by mpvader