victronenergy/venus #1217: Add maximum levels + capturing them for gui-v2

mpvader, 2024-02-07:
### todo:

- [ ] Jesus will make a list of parameters to be captured. Includes number of decimals per item.
- [ ] Qinetic to review that list.
- [ ] Rein will then implement capturing of maxima in Systemcalc, as well as having them available on D-Bus. 

### notes:
- this concerns both DC and AC gauges
- For AC, don't capture a value per phase. Max is the same for all phases.
- how to visualise AC gauges this on the brief page as well as the overview page is not yet clear that is on hold. DC gauges can be implemented.
- maxima are stored in local settings, which is also nice if some day in the future we have a feature that supports copying settings over from one GX to another.
- in the settings, there will be a list of all these gauges (and their maximum value). And on top of that two generic settings: first of all the mode (manual + automatic capturing; where as automatic is the default), and secondly an option to reset it all to zero.

Relates to this issue in gui-v2:
https://github.com/victronenergy/gui-v2/issues/532

fyi @izak

jepefe, 2024-02-23:
@ReinvdZee here the paths to monitor.

All needed paths for the gauges scaling are available in `com.victronenergy.system`

For AC IN and AC OUT we use current as it is easy to match to the current limit settings and and circuit breakers are rated in amps. The gauges must use the current reading for scaling also when the numerical value is displayed in watts.

For the rest of readings power is used because current is not available or possible to calculate.

### AC IN
Top left gauge
 
- Quattros have two AC inputs, and both need to be monitored separately.
- Minimum value must be monitored, negative values are reported when feeding to the grid.

This path reports the load of the active AC input
`com.victronenergy.system/Ac/ActiveIn/L1/Current  <- Decimals: 2`

This path tells which AC input is active. 
`/Ac/ActiveIn/ActiveInput  <- Active input: 0 = ACin-1, 1 = ACin-2, Decimals: 2`

### AC Output
Top right gauge

`Ac/ConsumptionOnOutput/L1/Current -> Decimals: 2
`

### DC Input
The current design states that when available, the DC input will be merged into one "power source" gauge in the Brief overview. This is not possible anymore, Serj is working on solving this.

@izak, can you help with this? We need the total DC generated power to show it in the same way we show AC In. We don't need PV power as we show it separately.

I see these paths in systemcalc code that can be added to get the value we need but I'm missing an equivalent path for com.victronenergy.dcsource.

`/Dc/Charger/Power + /Dc/FuelCell/Power + /Dc/Alternator/Power `

### DC Output gauge 
Bottom right gauge

Current is not available for DC system
`/Dc/System/Power -> Decimals: 0
`
### PV Power
Bottom left gauge

There are two possible sources PV Chargers and PV inverter.
Systemcalc already provides the total for all the PV Chargers.
PV Inverters can be in different positions and have multiple phases

We need to add PV charger and all the PV inverters to get the total PV power.
```
/Dc/Pv/Power -> Decimals: 0
+
/Ac/PvOnGrid/L1/Power -> Decimals: 0
/Ac/PvOnGrid/L2/Power -> Decimals: 0
/Ac/PvOnGrid/L3/Power -> Decimals: 0
+
/Ac/PvOnGenset/L1/Power -> Decimals: 0
/Ac/PvOnGenset/L2/Power -> Decimals: 0
/Ac/PvOnGenset/L3/Power -> Decimals: 0
+
/Ac/PvOnOutput/L1/Power -> Decimals: 0
/Ac/PvOnOutput/L2/Power -> Decimals: 0
/Ac/PvOnOutput/L3/Power -> Decimals: 0
```

mpvader, 2024-02-27:
@ReinvdZee pls prioritise this one.

ReinvdZee, 2024-02-27:
@jepefe:

>### AC IN
>Top left gauge
>- Quattros have two AC inputs, and both need to be monitored separately.
>- Minimum value must be monitored, negative values are reported when feeding to the grid.

Only minimum? we also need the maximum, right?

jepefe, 2024-02-27:
@ReinvdZee Yes, I assumed the maximum so I didn't mention it, both max and min are needed.

ReinvdZee, 2024-02-27:
> @ReinvdZee Yes, I assumed the maximum so I didn't mention it, both max and min are needed.

@jepefe, thanks.

kwindrem, 2024-02-27:
Power gauges should always have 0 as the minimum except of course for feed-in. If the minimum value is also scaling then this probably accounts for the twitchiness of the gauges when the value is small.

And I'll put my vote in for NEVER scaling a gauge. If there are no obvious limits or the installer has not provided the limits then the gauge should not be shown.

Imagine a car gas gauge that scales so that 1/8 of a tank is shown as it's maximum.

ReinvdZee, 2024-02-28:
@jepefe, @izak, I pushed an initial [implementation](https://github.com/victronenergy/dbus-systemcalc-py/commit/1ef998c2e9e0ecdd71547d0b463b3961064a9a7f). Please see below and let me know if you have any comments. 

> @ReinvdZee here the paths to monitor.
>
> All needed paths for the gauges scaling are available in `com.victronenergy.system`
>
> For AC IN and AC OUT we use current as it is easy to match to the current [...]

These paths are created for this:
 - '/Ac/In/1/Current/Min'
 - '/Ac/In/1/Current/Max'
 - '/Ac/In/2/Current/Min'
 - '/Ac/In/2/Current/Max'

`/Ac/In/1/Current/*` is updated when the active input is 1. `/Ac/In/2/Current/*` is updated when the active input is 2. I made sure that `/Ac/In/*/Current/Min` is 0 at its maximum to prevent a gauge starting at some arbitrary positive value. 
Currently, `/Ac/ActiveIn/ActiveInput` is used to determine which input is active, but I believe the correct path is `/Ac/ActiveIn/Source`, see [here](https://github.com/victronenergy/dbus-systemcalc-py/blob/master/dbus_systemcalc.py#L831)

> ### AC Output
> Top right gauge
> 
> `Ac/ConsumptionOnOutput/L1/Current -> Decimals: 2 `

This path is added:
`/Ac/ConsumptionOnOutput/Current/Max`

And contains the maximum of all three phases: max(`/Ac/ConsumptionOnOutput/[L1, L2, L3]/Current`)

> ### DC Input
> The current design states that when available, the DC input will be merged into one "power source" gauge in the Brief overview. This is not possible anymore, Serj is working on solving [...]

This path is added:

`/Dc/Input/Power/Max` 

And is the maximum of the sum of `/Dc/Charger/Power`, `/Dc/FuelCell/Power` and `/Dc/Alternator/Power`. 
It seems to be that there is still a path missing for `com.victronenergy.dcsource`. @izak?

> ### DC Output gauge
> Bottom right gauge
> 
> Current is not available for DC system `/Dc/System/Power -> Decimals: 0 `

This path is added:

`/Dc/System/Power/Max`

Which contains the maximum of `/Dc/System/Power`. 

> ### PV Power
> Bottom left gauge
>
> There are two possible sources PV Chargers and PV inverter. Systemcalc already provides the total for all the PV Chargers. PV Inverters can be in different positions [...]

This path is added:
`/Pv/Power/Max`

Which holds the maximum of the sum of all of the values listed above.

jepefe, 2024-03-06:
@ReinvdZee 
Please change for the AC Output gauge use:
`Ac/ConsumptionOnOutput/L1/Current`
+
`Ac/ConsumptionOnInput/L1/Current`

ReinvdZee, 2024-03-06:
Changed the script to use the max of `/Ac/Consumption/[L1, L2, L3]/Current` for the AC output gauge. The dbus path this maximum value is stored in is `/Ac/Consumption/Current/Max`

ReinvdZee, 2024-03-11:
@izak,

Can you do a short review of the [changes](https://github.com/victronenergy/dbus-systemcalc-py/commit/737be53ce03f8fb23fe3e98e3507ccc234cea541)? 

If all ok, Qinetic can use the min/max levels in gui-v2.

jepefe, 2024-03-14:
@ReinvdZee 
As discussed yesterday, AC consumption maximum value will differ whether the grid/generator is connected or not.

What we are going to do is monitor AC consumption maximum from the same source `ConsumptionOnInput  + ConsumptionOnOutput` but store it separately for these three scenarios.
- AC input 1 is connected: `Ac/In/0/Connected == 1`
- AC input 2 is connected: `Ac/In/1/Connected == 1`
- No AC input is connected. 

Above three cases are mutually exclusive.

Then `/Ac/Consumption/Current/Max` will report the corresponding maximum value.

When in manual mode the installer must be able to set the maximum value for each one of these cases instead of having only one maximum value for the AC consumption gauge.

ReinvdZee, 2024-03-18:
>What we are going to do is monitor AC consumption maximum from the same source ConsumptionOnInput  + ConsumptionOnOutput but store it separately for these three scenarios.

I've implemented the 3 scenarios. `dbus-systemcalc` maintains 3 maxima and relays the correct one to `/Ac/Consumption/Current/Max`. 

All min/max values are also synched to localsettings (stored under `/Settings/Gui/Gauges..`. I've also added a setting `/Settings/Gui/Gauges/Automax` to switch between automatically updating min/max values (synching from systemcalc to localsettings) or to use manually entered values (synching from localsettings to systemcalc). When set to 1 (default), the min/max values are automatically updated. 

The gui can take the min/max value of the gauges from the discussed paths, and should set the other limit to 0(lower limit for the `../Max` paths, upper limit for the `../Min` paths)

jepefe, 2024-03-28:
Yes, the power from the PV inverter is also accounted.

> Does PV inverter contribution factor into this as well?

mpvader, 2024-04-19:
@ReinvdZee , it sounds like these min and max values are on dbus now in two places: com.victronenergy.system as well as com.victronenergy.settings.

Why is that? Why not have it only in localsettings? And gui-v2 pages that need it to draw their gauges can take it from there?

ReinvdZee, 2024-04-22:
@mpvader, 
The idea was to be able to throttle the writes to localsettings if needed. But I suppose writing to localsettings once a second (systemcalc iteration) is not a problem.

Secondly, there is also a slight difference between the two in order to facilitate this:
>What we are going to do is monitor AC consumption maximum from the same source ConsumptionOnInput  + ConsumptionOnOutput but store it separately for these three scenarios.

>- AC input 1 is connected: Ac/In/0/Connected == 1
>- AC input 2 is connected: Ac/In/1/Connected == 1
>- No AC input is connected.

>Above three cases are mutually exclusive.
Then /Ac/Consumption/Current/Max will report the corresponding maximum value.

`com.victronenergy.system` contains only the path `/Ac/Consumption/Current/Max`, while `com.victronenergy.settings` contains paths for the three scenarios and relays the active one to `com.victronenergy.system` based on `/Ac/In/[0/1]/Connected`. Of course, gui-v2 can also do this check, or we also store `/Ac/Consumption/Current/Max` in localsettings.

So no, I don't actually have a compelling argument against using only the paths in localsettings. I guess it grew like this because the manual limits feature (and therefore the need to store settings) was introduced later.

mpvader, 2024-04-24:
Ok, Is this the conclusion @ReinvdZee ?

- systemcalc can be changed to no longer put min and max data on dbus. Only task for it is to, when autoranging is on, monitor values that it sees coming past and update the items in localsettings if the actual reading exceeds the min or max.
- gui-v2 uses the paths in localsettings only. When constructing the gauges as well as for its settings page.

or do you prefer it differently? I don’t mind either way, as long as its clear how it needs to be implemented in gui-v2 + there is a good reason for it + documented here: https://github.com/victronenergy/venus/wiki/dbus

ReinvdZee, 2024-04-24:
@mpvader,

>Ok, Is this the conclusion @ReinvdZee 

I agree this is the simplest implementation. I discussed this with Bea as well, and she is going to pull the values from localsettings anyway because she already has to set those paths when automax is off. So I guess the paths in `com.victronenergy.system` are useless then.

ReinvdZee, 2024-04-24:
@mpvader @jepefe

This actually raises some more questions; the manually entered limits are persistent through reboots, but they may be overwritten when automax is turned on. Right now, when automax is enabled after it's been disabled, the manually entered limits are updated only if a value higher than the set limit occurs.

Is this the desired behavior? Or do we want to clear the manually entered limits to have automax start over? Do we want to keep the manually entered limits elsewhere and restore them when automax is disabled again?

mpvader, 2024-04-24:
yes, how that works now it desired. And if in practice this turns out to be annoying, Bea can add a button to the settings page that resets all those limits to 0.

jepefe, 2024-04-24:
Keep in mind that switching to localsettings only solution will move some logic to the gui, for example pick the correct DBus path to use for de AC Output gauge depending on the currently active AC input.

jepefe, 2024-04-24:
Yes, I think that is best done in the gui rather than adding an extra setting.

ReinvdZee, 2024-04-24:
I pushed the [changes](https://github.com/victronenergy/dbus-systemcalc-py/tree/rein/max_level_gauges) and placed it on the todo.

closed 2024-04-30 by jhofstee