victronenergy/venus #1219: Venus for Rasberry Pi 5

espenairmine, 2024-02-08:
We'd like to try running Venus on Pi 5, any plans to build an image for the '5 in the next release(s)?

https://community.victronenergy.com/questions/240780/raspberry-pi-5-supprt.html

zxrjimmy1, 2024-07-26:
would aslo like to have it running on my pi 5

apharcha, 2024-08-11:
I finally ordered a Raspberry PI 4-b to be exact off of Amazon in last few days...  Now that Prices dropped to earth. Plan to upgrade my RPI-2 to 4 for VRM monitoring for Solar/Victron Inverter Quattro (2x 48V/3000/35). However, when I got the item in mail the box label sticker says "Raspberry PI 4-b 4GB" but on boot I see it's completely a RPI-5 with 2x mini-HDMI Ports and 8GB RAM. The red box reads Raspberry PI-5. Anyways, My dilema is whether I send it back or figure out how to get it running with VenusOS/RPI. 

Also, Is there a quick & dirty way I can possibly get RPI-5 by installing perhaps a Debian build on SDCard and then copy partial Boot/Driver interfaces from there to the the SD Card built for VenusOS / RPI 4-b? Not sure what I can do.

apharcha, 2024-08-11:
> would aslo like to have it running on my pi 5

did you get any luck on this post?

nmbath, 2024-08-11:
@apharcha There are certainly things you can do to rebuild u-boot and kernel and manually replace them on a VenusOS SDCard. However having done the work to get the RPI4 working on VenusOS they are fairly tightly integrated and fixing one issue can break others. The RPI4 had fixes in a number of upstream branches to make it work. 

There is no activity that I am aware of to get the RPI5 working, As you say now the prices are coming down and availability is better I may consider it over the winter months if there is nothing more important.

ortegafernando, 2024-12-11:
Hi, any expected date for this? Thanks

rdavis852, 2024-12-19:
Just adding my 2 cents.    Would be very grateful a  Pi-5 version.

penguinator13, 2025-02-23:
Would also really appreciate this! Anyone have any luck? Thanks.

geoffreybosch, 2025-05-22:
Seems like people with much more knowledge of the project seem to think this is more difficult than it seems. Is there anything the community can do to get this rolling?

nmbath, 2025-05-23:
There are a lot of low level changes, such as kennel and boot methodology (u-boot). All needing newer versions to be built. and then integrated into the Openembedded build system.

While it is great to have the latest and greatest hardware, there is little need to driver this change and thus motivation. A dual core RPI4 hss plenty of capacity to run VenusOS large and signalk/nodred. If you then start wanting more capabilities to utilise the CPU power, then VenusOS is not the way to go and interfacing those tools with a lower specification RPI that is supported will work. Tools such as dBus over IP, MQQT, etc can allow this.

psm12, 2025-05-23:
Mark,Many thanks for taking the time to explain. Paul