victronenergy/venus #1222: look into changing flashmq config without restarting it

mpvader, 2024-02-14:
This came up during discussion between Wiebe and Matthijs as an improvement.

We noticed that when changing MQTT service enabled/disabled settings in gui-v2 wasm, the UI got half-stuck: browsing around it still worked ofcourse, but all MQTT comms was halted. 

Until we refreshed the browser.

And the assumption is that this is caused by FlashMQ being restarted.

At which point Wiebe

wiebeytec, 2024-03-15:
FlashMQ is restarted by Venus Platform on change of 'VRM settings -> two way communication'.

I have proof of concept code done for dynamically adding and removing bridges without the need for a restart. _Changing_ the bridge config (usernames, password, host, etc) is a different story, but simply adding and removing should be possible soon.

That means venus-patform needs to be adjusted to not restart but reload FlashMQ when already running. 

Also, I think `/usr/sbin/mosquitto_bridge_registrator.py` doesn't have to issue a FlashMQ restart anymore. There is some code to detect changes in `/data/conf/flashmq.d/vrm_bridge.conf` and then restart, but generation of that file should be idempotent and atomic nowadays: in otherwords: fully correct or empty. But, I would to discuss that with @jhofstee .

wiebeytec, 2024-05-03:
FlashMQ 1.12.0 with hot-reloadable bridges has been released. 

Now Venus can be changed to not restart FlashMQ when 'two way communication' (or the yet-to-be-created 'VRM full access' setting). As far as I know, there currently is no support in Venus Platform to signal reload to running deamons based on config changes. So, some discussion may be required.

mpvader, 2024-09-28:
@jhofstee / @wiebeytec  can this be closed?

wiebeytec, 2024-09-30:
Yes, this can be closed. I'll close it.

There is one note:  `mosquitto_bridge_registrator.py` generates a different configuration when you have read-only mode or full mode, but it only writes this config to disk when it can successfully register at VRM. The reason is probably mostly historical, but also too intricate to fiddle with. 

So, when you change between full and read-only when you don't have internet, the config file `/data/conf/flashmq.d/vrm_bridge.conf` may not reflect the settings. This is a minor issue.

closed 2024-09-30 by wiebeytec